serialport/node-serialport #2246: Will not install after updating to Mac OS 11.3.1

ricozinn, 2021-05-13:
### Summary of Problem

I'm trying to install modbus-stream (which has serialport as a dependency) to an electronJS app.    Prior to updating to Mac OS 11.3.1 I had installed it and it was working great with my same package.json etc.  Now though I get a long list of errors.

I have read through your guide-installation instructions and even tried the electron-rebuild changes.  I've also tried updating to the latest Node and npm and I've confirmed I have python2.  I don't see anyone else reporting this issue so hopefully it is something easy that I missed, but like I said it was running last week and the only change I made was install the latest Mac OS update.

Note: I can run `npm install serialport` and install the latest serialport version 9.x but the modbus-stream requires an older build (v6.2.2).  

### Code to Reproduce the Issue
Here's my package.json
```
{
  "name": "ABC",
  "version": "1.0.0",
  "description": "ABC",
  "main": "main.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1",
    "start": "electron .",
    "pack": "electron-builder --dir",
    "dist": "electron-builder",
    "install": "electron-rebuild"
  },
  "build": {
    "appId": "abc.governor",
    "mac": {
      "category": "your.app.category.type"
    }
  },
  "author": "Richard Zinn",
  "license": "ISC",
  "devDependencies": {
    "electron": "^9.2.0",
    "electron-builder": "^22.8.0",
    "electron-rebuild": "^2.3.5"
  },
  "dependencies": {
    "modbus-stream": "^0.46.0"
  }
}
```

### Versions, Operating System and Hardware

- SerialPort@?6.2.2
- Node.js v?16.1.0
- Windows? Linux? Mac? Mac 11.3.1
- Hardware and chipset? (Prolific/FTDI/Other) N/A

Thanks in advance for any help,
Richard

Julusian, 2021-05-26:
When you updated macos did you also update node to v16?
The error is because serialport older than v9 does not support node 14 or newer. It may compile but as you are seeing may not.
The solution here is to either use a newer version of serialport (which means chasing the other lib to update the version they use), or use an older version of nodejs

GazHank, 2021-07-05:
Hi @ricozinn did updating to more recent versions resolve this issue for you?

ricozinn, 2021-07-12:
@GazHank I had already switched to modbus-serial, by the time this tip from @Julusian had come in and that module didn't have this problem.  So, can't help you out on that one sorry.

closed 2021-07-12 by ricozinn