serialport/node-serialport #2250: Worker Threads Crash

devwoojin, 2021-05-24:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I want to open a serial connection from within a WorkerThread.

- What happens?
I get this Error:

FATAL ERROR: HandleScope::HandleScope Entering the V8 API without proper locking in place
 1: 00007FF7B80B058F napi_wrap+109311
 2: 00007FF7B80552B6 v8::internal::OrderedHashTable<v8::internal::OrderedHashSet,1>::NumberOfElementsOffset+33302
 3: 00007FF7B8056086 node::OnFatalError+294
 4: 00007FF7B891305B v8::HandleScope::Initialize+107
 5: 00007FF7B88F8E8E v8::HandleScope::HandleScope+14
 6: 00007FF7B80D3D8D node::CallbackScope::CallbackScope+429
 7: 00007FF7B80D43E6 node::CallbackScope::~CallbackScope+1126
 8: 00007FF7B80D4A61 node::MakeCallback+209
 9: 00007FFE8FF1B2F6

- What should have happened?
SerialPort should work as normal.

### Code to Reproduce the Issue

main.js
```js
const { Worker, isMainThread } = require('worker_threads');

const worker = new Worker('./worker.js');
console.log('isMainThread:', isMainThread); // true

worker.on('message', (msg) => {
  console.log(msg);
});

// Code
```

worker.js
```js
const { Worker, isMainThread, parentPort, workerData } = require('worker_threads');
const SerialPort = require('serialport');

console.log("Serial Worker");

const port = new SerialPort("COM3", {baudRate: 115200}, function (err) {
  if (err) {
    console.log('Error: ', err.message)
    return;      
  }
});

port.on('data', function (data) {
  console.log('Data:', data)
})
  
// Code
```

### Versions, Operating System and Hardware

- SerialPort@?
 serialport": "^9.0.7",
- Node.js v?
v14.15.5
- Windows? Linux? Mac?
windows 10
- Hardware and chipset? (Prolific/FTDI/Other)
FTDI

reconbot, 2021-05-24:
We don't support workers at this time https://github.com/serialport/node-serialport/issues/1938

closed 2021-05-24 by reconbot

ScreamZ, 2023-11-15:
@reconbot Any recommendations on high-consuming serial communication (just like DMX) ? to avoid blocking the loop. Currently, I'm using https://github.com/moritzruth/node-enttec-open-dmx-usb/tree/main and it plays with micro sleep. Which could be nice to be used in another thread ?