serialport/node-serialport #2251: I cannot install serialport NODE_MODULE_VERSION error.

Mrolcraft, 2021-05-25:
Hi! I wanted to install and use serialport to use an arduino with NodeJS and electron. But after a lot of try, I cannot install serialport. After many issues, I have this issue and i find nothing to resolve it. I tried to use electron-rebuild but without success. I use nodejs version 14.17, and I'm on windows 10. Thanks a lot for your help !

```
D:\Developpement\UFC\node_modules\@serialport\bindings\build\Release\bindings.node' was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 89. Please try re-compiling or re-installing
```

reconbot, 2021-05-26:
How are you using electron rebuild?

devrim-oguz, 2021-05-27:
> Hi! I wanted to install and use serialport to use an arduino with NodeJS and electron. But after a lot of try, I cannot install serialport. After many issues, I have this issue and i find nothing to [...]

Hello, I faced with the same problem. The solution to this is to install the electron-rebuild package via npm and use it to rebuild the node modules. That solved the problem.

partap, 2021-06-30:
I'm having the same problem... I always need to run electron-rebuild and compile from source.  This is a pain because I build my app for macos, linux and windows and I need to build each release on the native platform.

It didn't used to be this way... For at least a couple years, I was able to use electron-builder on macos to build my linux and windows releases.  It would automatically download and install the proper prebuilt bindings.  Then it stopped working for windows... But then recently it stopped working for linux and, indeed, even locally on macos.  

Now it seems I *always* need to recompile from source on the native platform.  For some reason it doesn't download prebuilt bindings, even when they appear to be available on github...not sure if this is a problem with electron-builder or the serialport repo.

I miss being able to deploy for 3 platforms with a single command, rather than having to spin up VMs for each :(

GazHank, 2021-06-30:
> Now it seems I _always_ need to recompile from source on the native platform. For some reason it doesn't download prebuilt bindings, even when they appear to be available on github...not sure if this is a problem with electron-builder or the serialport repo.

@partap Is it possible that you have your project.json configured to build from source? `"buildDependenciesFromSource": true,"npmRebuild": false,`

I know I've previously forgotten to remove those settings after the prebuilds are updated

seriouz, 2021-07-09:
I have the exact same problem with Node 14 and Electron 13:
```
no serial port found: Error: The module '\\?\C:\code\appviewer2\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 89. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

GazHank, 2021-07-09:
Hi @seriouz

It looks like you need to update your installed version of Serialport after upgrading your version of electron

The latest version of serialport includes prebuilds for electron 13 (v89) per https://github.com/serialport/node-serialport/releases/tag/%40serialport%2Fbindings%409.2.0

If you upgraded serialport to v9.0.2 first and then subsequently installed electron 13, yarn or non might not bother to redownload the correct binary. If that is the case then it might be simplest to clear your node modules directory and do a full reinstall of your modules to clean things up.

seriouz, 2021-07-09:
Thanks for the reply.
Putting the file manually into the node_modules folder worked. Getting it downloaded through npm or yarn did not work. Even without an exiting node_modules folder.

GazHank, 2021-07-09:
Thanks @seriouz I'm glad to get it to work.  Were you getting an error during the install, or just issues with the package manager downloading the wrong version?

seriouz, 2021-07-09:
I just tried again and i saw no error during the installation process `yarn install`. I have the latest NodeJs version, the latest Electron and the latest serialport version. Still got the `NODE_MODULE_VERSION` error. Seems that the wrong version of serialport is downloaded.

GazHank, 2021-07-09:
And each time you get 'compiled against a different Node.js version using
NODE_MODULE_VERSION 83'?  (Node version 14) That is a little strange.  I've seen instances where the version downloaded is based on node version rather than electron version, or with the install not overriding previous installs, but this sounds a little unusual if you are using node 16 (NODE_MODULE_VERSION 93).  I'll need to see if there is a way I can recreate the problem

seriouz, 2021-07-09:
Sorry just to clearify: I have the NodeJS Version 14.17.3 (Latest LTS) and Electron 13.1.6 and Serialport 9.2.0.

GazHank, 2021-07-09:
Ah that makes more sense, it looks like the install is using your node version rather than your electron version.  It should only be using your electron version (when used in an electron app).  I'm away from my PC at the moment, but will double check how to fix it when I get back

GazHank, 2021-07-10:
I think that I might need to tweak the electron installation instructions a little, to recommend the use of electron rebuild rather than suggesting it as optional.

In referring to the [electron docs](https://www.electronjs.org/docs/tutorial/using-native-node-modules#troubleshooting), their immediate solution is to use electron-rebuild, and this is exactly what we use in the [electron-serialport example](https://github.com/serialport/electron-serialport).  This way the installation uses your version of electron rather than your version of node for the package installation.  To do this they suggest:

```
npm install --save-dev electron-rebuild

# Every time you run "npm install", run this:
./node_modules/.bin/electron-rebuild

# If you have trouble on Windows, try:
.\node_modules\.bin\electron-rebuild.cmd
```
A quick way to check if this would work for you would be to try the [electron-serailport example](https://github.com/serialport/electron-serialport), the [instructions can be found here](https://github.com/serialport/electron-serialport#to-use).  To handle the install -> rebuild we simply add it into the [package scripts](https://github.com/serialport/electron-serialport/blob/3f04b2d35299bf8026c4469f0d3deb2da6512ee1/package.json#L6-L9)

There are some [alternative instructions which would change your npm environment variables](https://www.electronjs.org/docs/tutorial/using-native-node-modules#using-npm), but that would make it messy if you have multiple packages, some of which don't use electron, and it means that any one else who wants to use your project would need to change the config.

@seriouz would you be able to check if electron-rebuild works for you, and if so I'll update the docs for electron to clarify this

seriouz, 2021-07-10:
Executing `.\node_modules\.bin\electron-rebuild.cmd` did the trick! Thank you :)

GazHank, 2021-07-10:
Thanks for the speedy response @seriouz I've updated the docs to clarify a few steps, but will also add that windows specific piece since that helped you

If anyone else is still experiencing issues then please let me know, otherwise I'll update that additional tweak to the  docs and close this issue

GazHank, 2021-07-10:
https://github.com/serialport/website/pull/56 should address the additional windows specific info, but if there are any other changes to the [electron docs](https://serialport.io/docs/guide-installation#electron) that would help please let us know. Any suggestions would be very welcome

closed 2021-07-10 by GazHank

hdpklm, 2021-09-06:
> I think that I might need to tweak the electron installation instructions a little, to recommend the use of electron rebuild rather than suggesting it as optional.
>
> In referring to the [electron [...]

Hi @GazHank
I run the command `.\node_modules\.bin\electron-rebuild.cmd` in windows 10. and fix the error of NODE_MODULE_VERSION.
but the problem is when I use the serialport module in vscode extension and run any command of this extension, the vscode is closed imediatly without any comment, any error, or leave any message.

thank you a lot