serialport/node-serialport #2252: UART with Raspberry Pi Zero only receiving 4 Bytes

oliverwerlen, 2021-05-26:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I want to use serial port to communicate via UART with a raspberry pi. I need to write and to read. 
- What happens?
Writing works fine, but when I want to read the Data, I only get the first 4 Bytes. 
- What should have happened?
There are sended way more than the received 4 Bytes (more or less about 20)

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

// Code
```
const SerialPort = require('serialport')
const port = new SerialPort('/dev/ttyS0', {
  baudRate: 9600
})

// Read data that is available but keep the stream in "paused mode"
port.on('readable', function () {
  console.log('Data:', port.read())
})

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v? 14.13.0
- Windows? Linux? Mac? Linux (rasbian)
- Hardware and chipset? (Prolific/FTDI/Other)  Raspberry Pi Zero W

reconbot, 2021-05-26:
You have to keep calling read until it returns null, and then wait for another readable event. Or treat the stream like an async iterator which is what I'm guessing you'd prefer?

The reason here is the data doesn't arrive all at once and as soon as any data arrives the readable event fires.

oliverwerlen, 2021-05-26:
Yes, I would prefer your second solution, with the async iterator. 
Ok, I understand the problem now way better. 

Can you give me a code example of the async iterator?

tsofist, 2022-02-05:
> Yes, I would prefer your second solution, with the async iterator. Ok, I understand the problem now way better.
> 
> Can you give me a code example of the async iterator?

Hi! Did you succeed in finding an example?