serialport/node-serialport #2255: Cant install on raspberry pi

sandermm, 2021-05-30:
### Summary of Problem

Cant figure out what's the problem (worked fine on my previous installation). I was trying to install serial, but it keeps giving this output:

root@DietPi:~# sudo npm install serialport --unsafe-perm --build-from-source

> @serialport/bindings@9.1.0 install /root/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

gyp ERR! build error 
gyp ERR! stack Error: not found: make
gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/which/which.js:13:12)
gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:68:19)
gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:80:29)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:89:16
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
gyp ERR! stack     at FSReqCallback.oncomplete (fs.js:192:21)
gyp ERR! System Linux 5.10.17-v8+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/node_modules/@serialport/bindings
gyp ERR! node -v v14.17.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/root/package.json'
npm WARN root No description
npm WARN root No repository field.
npm WARN root No README data
npm WARN root No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.1.0 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@9.1.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-05-30T17_03_33_794Z-debug.log

### Versions, Operating System and Hardware

- SerialPort@9.1.0
- Node.js v14.17.0
- NPM 6.14.13
- Linux - DietPi v7.2 64 bit
- Raspberry Pi 4

reconbot, 2021-05-30:
You're missing some required packages. https://serialport.io/docs/guide-installation#raspberry-pi-linux

sandermm, 2021-05-30:
Yes, but wich ones? I got Python 2.7.16 and node-gyp, where can I get a list with requirements?

reconbot, 2021-05-30:
I don't known anything about DietPi, but you're missing the gcc build tools. The error message complains about cmake.

sandermm, 2021-05-31:
Ah that was it! Thank you very much!

reconbot, 2021-05-31:
World you be able to enhance our rpi docs a bit?

sandermm, 2021-06-03:
Yes I can try, what do you need?

reconbot, 2021-06-03:
Well, whatever would have helped you.