serialport/node-serialport #2259: SerialPort.list() crashes Node when used with brother device

rb-serin, 2021-06-03:
### SerialPort.list() crashes Node ( used on electron) when a Brother printer is attached

(Please answer all 3)

- What are you trying to do?
Listing all devices attached to the machine

- What happens?
Node crashes leaving a blank page on electron renderer process

- What should have happened?
Getting a list of devices or an exception

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')
SerialPort.list()

```

### Versions, Operating System and Hardware

- SerialPort@ 7.x and 9.x 
- Node.js v14.15.4
- Windows? Linux? Mac? Windows
- Hardware and chipset? (Prolific/FTDI/Other) Intel core i5 with Brother printer attached

This is the same problem as #1518
See also https://www.gitmemory.com/issue/serialport/node-serialport/2155/771687008

GazHank, 2021-06-07:
Hi @rb-serin

Would you be able to confirm the specific version(s) of electron and serial port that you are using?

Also, could you confirm if you are facing the issue of the device not being assigned a COM port per the prior issue you referenced?

If the COM port isn't being assigned by windows you might want to manually configure the COM port within device manager (there are some instructions here: https://www.virtual-serial-port.org/articles/com-port-on-windows-10/#assign ).  Although even if that does resolve your issue it sounds like the error handling/ fault tolerance could be improved.

GazHank, 2021-07-05:
This might be related to https://github.com/serialport/node-serialport/issues/2155 as both seem to relate to unexpected behaviour; in both cases I think we need more info to be able to investigate and debug

nyholmjan, 2021-08-25:
I can reliably reproduce this issue. The key part is to query for ports while windows is installing usb to serial drivers. It can be reproduced simply by removing the usb to serial drivers of the device from device manager and then unplugging the device and plugging it in again, the crash will happen when serialport.list() is called while the drivers are being reinstalled.

## Versions, Operating System and Hardware
- serialport: 9.2.0
- electron: 13.2.1
- os: windows 10
- hardware: u-blox EVK-M101

## Code to reproduce
```
const serialport = require("serialport");
window.setInterval(() => {
    serialport.list().then(ports => {
        console.log(ports)
}, 1000);
```

GazHank, 2021-09-06:
I wonder if this is due to changes being made to the data stored in the windows registry as we are executing the list().  Since much of the data returned by the list operation comes from the registry if it's content changes mid process it could causes this breakage.  I can't see any unhandled exceptions based on a quick scan of the code, but that does feel like the most likely issue.

I had previously thought about avoiding direct registry access, and using the windows APIs more fully to provide us a little more abstraction, in the hope that the APIs would provide better fault tollerance and handling of weird edge cases.  I'll try to take a look at this again when I get some time.  If anyone wants to take a look at the code in question and perform some debugging, the EIO_List and associated methods it calls can be found @ https://github.com/serialport/node-serialport/blob/master/packages/bindings/src/serialport_win.cpp#L782

pseudorandomseed, 2021-09-28:
I can confirm that this is an issue specifically with Brother printers connected via USB. I began building with SerialPort back in 2017 and I can also confirm that this issue has been ongoing since I started with v5.0.0. My app is installed on hundreds of computers, and every single customer who has a Brother printer connected via USB has the same issue: the app crashes as soon as it tries to list serial ports while a Brother printer is plugged in via USB.

I have had this issue with both Windows 7 and Windows 10 on both x86 and x64 machines. I have followed all of the updates over the years, and I am currently running SerialPort v9.2.3 on NWJS v0.56.1 (node v16.9.1). When this happens, a crashdump file is generated and I've opened it up in WinDbg (though interpreting this output is well beyond my skillset). The output from WinDbg can be seen here: https://gist.github.com/pseudorandomseed/b5fefb85ea235c6d8f9a5370f4fcc83c.

I've tried everything to handle any uncaught exceptions, etc., to no avail as the app completely crashes as soon as it attempts to list the serial ports. If anyone thinks they might be able to help get to the bottom of this, I am happy to work with you and provide additional details or try different things on my end as I have everything needed to consistently reproduce this problem and I would love to see it finally get resolved.

GazHank, 2021-09-28:
thanks @pseudorandomseed that's great news!

I mean that you have access to a device that exhibits this issues... I'm really sorry that this is causing crashes for you (and your customers).

Would you be able to try adding print statements to the EIO_List method per @reconbot suggestion: https://github.com/serialport/node-serialport/issues/2155#issuecomment-783834294

This would allow us to see at which step of the process the crash is occurring?  (let me know if you would like any help or support to make the changes, or to setup your windows machine to be able to build the code)

pseudorandomseed, 2021-09-28:
> thanks @pseudorandomseed that's great news!
> 
> I mean that you have access to a device that exhibits this issues... I'm really sorry that this is causing crashes for you (and your customers).
> 
> Would you be able to try adding print statements to the EIO_List method per @reconbot suggestion: [#2155 (comment)](https://github.com/serialport/node-serialport/issues/2155#issuecomment-783834294)
> 
> This would allow us to see at which step of the process the crash is occurring? (let me know if you would like any help or support to make the changes, or to setup your windows machine to be able to build the code)

If someone could guide me on how to do it, or better yet, provide a modified https://github.com/serialport/node-serialport/blob/master/packages/bindings/src/serialport_win.cpp#L782, I would be happy to do so. The only way I can get SerialPort to build for NWJS (nw-gyp) is with Visual Studio 2015 and Python 2.7, so those are the tools I have installed, but I'm happy to set up anything else that might be needed. Would I just put this new .cpp in the bindings folder and rebuild with nw-gyp?

reconbot, 2021-09-28:
That's about right

caffeinatedbits, 2021-09-29:
@reconbot see https://github.com/serialport/node-serialport/pull/2325

gabrieleimotion, 2021-10-21:
I have the same issue where on Windows the node app (no electron involved) crashes when getting list of the devices from time to time. A brother label printer ql-820 is the problem, currently using the node-printer library to make it work (meanwhile othger serial ports are used for other purposes)

The app crashes without exceptions, how can I collect a trace of that?

reconbot, 2021-12-12:
Can someone test `serialport@10.0.0` to make sure this is still good?

nyholmjan, 2021-12-15:
> Can someone test `serialport@10.0.0` to make sure this is still good?

It hasn't been verified that the similar issue with the u-blox devices described above has the same cause, but I have tested that one with serialport@10.0.0 and the issue still persists.

GazHank, 2021-12-16:
> > Can someone test `serialport@10.0.0` to make sure this is still good?
> 
> 
> 
> It hasn't been verified that the similar issue with the u-blox devices described above has the same cause, but I have tested that one with serialport@10.0.0 and the issue still persists.

Are you also experiencing the u-blox device issue with the latest v9 release too?  If so I wonder if we have a different defect we need to address for that device

closed 2022-04-28 by stale[bot]