serialport/node-serialport #2260: UnhandledPromiseRejectionWarning when trying to catch open of non-existant port

petertorelli, 2021-06-07:
### Summary of Problem

- What are you trying to do? Open a port that does not exist.
- What happens? Fail to catch an unhandled promise rejection.
- What should have happened? Catch should have caught the error.

It could be the fact that it is early Monday morning and I haven't messed with `serialport` in a year, but with the latest version, if new(+open) should throw...

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport');
(async () =>{
	try {
		const x = new SerialPort('noportbythisname', { autoOpen: false });
		x.open();
	} catch (error) {
		console.error(error);
	}
})();
```
```
% node test.js 
(node:47978) UnhandledPromiseRejectionWarning: Error: Error: No such file or directory, cannot open noportbythisname
(node:47978) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:47978) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

% node -v
v12.14.0

% grep serialport package.json 
    "serialport": "^9.1.0"
```

### Versions, Operating System and Hardware

- SerialPort@9.1.0
- Node.js@12.14.0
- Mac Big Sur
- Hardware and chipset?  STLink V2

petertorelli, 2021-06-07:
Also, I thought I was forgetting how await/async worked, so I started with the basic:

```js
const SerialPort = require('serialport');
try {
	let port = new ("noportbythisname", { autoOpen: false });
	port.open();
} catch (error) {
	console.error(error);
}
```

petertorelli, 2021-06-07:
Just realized this is an unhandled emitter error. I think the documentation is a bit misleading: open doesn't throw, it emits an error.

closed 2021-06-07 by petertorelli

reconbot, 2021-06-07:
Happy to merge a note if you want to edit the docs https://serialport.io/docs/api-stream#serialportopen