serialport/node-serialport #2263: queries sent to the serial port but not heard, being repeated in parser.once("data", function (resp) {}) when communication occurs

ana-marcia, 2021-06-10:
### Summary of Problem

- What are you trying to do?
I use FTDI communication with a device. I send questions to serial every 3000ms. Sometimes the device is not connected to FTDI, that is, I don't get any response.

- What happens?
But when I connect the equipment again, all writings still unanswered are answered, this way I replicate the data the amount of times that were written unanswered.

- What should have happened?
I need that every time the writing is unanswered, it is not broadcast or heard again. I tried using flush() but it didn't work for me

### Code to Reproduce the Issue

```js
  port.flush(() => {
          port.write(getPack(0, _ts.value), function (err) {
            _ts.notData = true;
            _ts.countDownTimer();
            if (err) {
              console.log("ERR: " + err);
            }
          });
          port.drain(() => {
            console.log("write done");
            
          });
        });

parser.once("data", function (resp) {
          let data = resp.toString("utf8");
          if (data) {
            console.log(data);
           }
})

// Code
```

### Versions, Operating System and Hardware

- SerialPort@? **^9.0.0**
- Node.js v? **v14.15.5**
- Windows? Linux? Mac? **Windows**
- Hardware and chipset? (Prolific/FTDI/Other) **FTDI**

GazHank, 2021-06-14:
@ana-marcia I'm not sure if the issue stems from how the flush and drain function calls are interacting.  I'd expect you to call a write and then either flush it or drain it.  I think that the drain function is causing the system to indefinately wait until the data is transfered irrespective of the flush request.

Since you are using a the drain function I wonder if you want to add some sort of timeout for it?
This might not be ideal for you, but one of my projects uses: 
```
async function read_cb(callback) {
  var time = 0;

  var timeout = function () {
    setTimeout(function () {
      time += 50;
      focus._port.drain(err => {
        if (err) {
          if (time > MAX_MS) {
            callback(true, "TIMED OUT");
          }
          timeout();
        } else {
          callback(null, "drain");
        }
      });
    }, 50);
  };
  timeout();
}
```

@reconbot should the [doc for serialport flush](https://serialport.io/docs/api-stream#serialportflush) be update to reflect that on Windows  [PurgeComm](https://docs.microsoft.com/en-us/windows/win32/api/winbase/nf-winbase-purgecomm) is used instead of FlushFileBuffers?  (FlushFileBuffers only seems to be used for the Drain function)

https://github.com/serialport/node-serialport/blob/0461285f04b93034008cf6c22ac566d0a1a09571/packages/bindings/src/serialport_win.cpp#L936-L944

Happy to propose a PR to the docs if that would help

reconbot, 2021-06-14:
Sounds reasonable

closed 2022-04-28 by stale[bot]