serialport/node-serialport #2268: bindings-v9.1.0-electron-v89-win32-x64.tar.gz is not found and giving 404 while yarn install

AtiqGauri, 2021-06-15:
## 💥 Proposal

### What feature you'd like to see
Please add new binaries for electron v89 win32/darwin 

## Error 
To ensure your native dependencies are always matched electron version, simply add script `"postinstall": "electron-builder install-app-deps" to your `package.json`
  • rebuilding native dependencies  dependencies=@serialport/bindings@9.1.0 platform=darwin arch=x64
  • install prebuilt binary  name=@serialport/bindings version=9.1.0 platform=darwin arch=x64 napi=
  • build native dependency from sources  name=@serialport/bindings
                                          version=9.1.0
                                          platform=darwin
                                          arch=x64
                                          napi=
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 6.1.2
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for local prebuild @ prebuilds/bindings-v9.1.0-electron-v89-darwin-x64.tar.gz
    prebuild-install info looking for cached prebuild @ /Users/runner/.npm/_prebuilds/5eb8b0-bindings-v9.1.0-electron-v89-darwin-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.1.0/bindings-v9.1.0-electron-v89-darwin-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.1.0/bindings-v9.1.0-electron-v89-darwin-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=13.1.2 runtime=electron arch=x64 libc= platform=darwin)
    
  • rebuilding native dependency  name=@serialport/bindings version=9.1.0
  • packaging       platform=darwin arch=x64 electron=13.1.2 appOutDir=dist/mac
  • downloading     url=https://github.com/electron/electron/releases/download/v13.1.2/electron-v13.1.2-darwin-x64.zip size=79 MB parts=6
  • downloaded      url=https://github.com/electron/electron/releases/download/v13.1.2/electron-v13.1.2-darwin-x64.zip duration=822ms
  • signing         file=dist/mac/mutesync.app identityName=Developer ID Application: Karanbir Toor (FMTR4QU934) identityHash=6DD5CB34B841EA89A2214A9A2D161824BCA5E402 provisioningProfile=none
  • building        target=macOS zip arch=x64 file=dist/mutesync-4.5.2-TEST-mac.zip
  • building        target=DMG arch=x64 file=dist/mutesync-4.5.2-TEST.dmg
  • rebuilding native dependencies  dependencies=@serialport/bindings@9.1.0 platform=win32 arch=x64
  • install prebuilt binary  name=@serialport/bindings version=9.1.0 platform=win32 arch=x64 napi=
  ⨯ cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible
                                    cause=exit status 1
                                    errorOut=prebuild-install info begin Prebuild-install version 6.1.2
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for local prebuild @ prebuilds/bindings-v9.1.0-electron-v89-win32-x64.tar.gz
    prebuild-install info looking for cached prebuild @ /Users/runner/.npm/_prebuilds/473210-bindings-v9.1.0-electron-v89-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.1.0/bindings-v9.1.0-electron-v89-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.1.0/bindings-v9.1.0-electron-v89-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=13.1.2 runtime=electron arch=x64 libc= platform=win32)
    
                                    command=/Users/runner/hostedtoolcache/node/16.3.0/x64/bin/node /Users/runner/work/mutesync/mutesync/shells/electron/node_modules/prebuild-install/bin.js --platform=win32 --arch=x64 --target=13.1.2 --runtime=electron --verbose --force
                                    workingDir=/Users/runner/work/mutesync/mutesync/shells/electron/node_modules/@serialport/bindings
  • Above command failed, retrying 5 more times
  • Above command failed, retrying 4 more times
  • Above command failed, retrying 3 more times
  • Above command failed, retrying 2 more times
  • Above command failed, retrying 1 more times
  • Above command failed, retrying 0 more times
  • building embedded block map  file=dist/mutesync-4.5.2-TEST-mac.zip
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
Error: Process completed with exit code 1.

GazHank, 2021-06-15:
@AtiqGauri, while waiting on the pre-build becoming available, you might want to try using the electron-rebuild tool.  The instructions were recently updated @ https://serialport.io/docs/guide-installation#electron

If you have any issues or suggestions for inprovements to the instructions then please shout.

hopefully a prebuild will be available soon :-)

closed 2021-06-18 by reconbot

reconbot, 2021-06-19:
Please try `serialport@9.2.0`

PR #2274 (2021-06-18): fix: electron 13 prebuilds
Simply updates node-abi

This will make the next release produce prebuilds for electron 13, electron 12 and node 16

fixes #2234 fixes #2268