serialport/node-serialport #2269: Maintainers

reconbot, 2021-06-15:
Hey @GazHank you've been really helpful lately and I really appreciate it. I'd love to meet you but I don't have any way to get in contact with you.

-Francis

GazHank, 2021-06-15:
Hey Francis, I'm just trying to help out, when I can.

One of my projects was depending on node-serialport for ages before I even knew about it.  My first real sight of the project was when something broke, and I had to dig around to see what had happened.  I see the great stuff you are doing, and just want to lend a hand if I can.  I'm hoping that on balance I'm helping rather than hindering.

If you would like to get in touch you can reach me at gazhank@gmail.com

reconbot, 2021-06-15:
You're helping a ton, I really do appreciate it!

closed 2021-06-15 by reconbot