serialport/node-serialport #2270: Enhancement: change version inter dependencies

GazHank, 2021-06-15:
## 💥 Proposal

### What feature you'd like to see
Change dependencies of the serialport package to use exact versions. 

As discussed in https://github.com/serialport/node-serialport/issues/2266

@GazHank 
> @reconbot would it be prudent for us to change the package.json for the serialport package to be exact? That is to say serialport 9.0.0 would have a dependency on "@serialport/bindings": "9.0.0" instead of "@serialport/bindings": "^9.0.0"

Seems reasonable, we need to see how lerna supports it.

__Originally posted by @reconbot in https://github.com/serialport/node-serialport/issues/2266#issuecomment-861603622__

## Motivation
Simply troubleshooting of issues introduced in sub-packages. 

Avoid having to use resolutions or for projects to revert to earlier major versions.

## Pitch

At present if someone has a dependency on @serialport 9.0.0 then it creates a dependency on the sub packages as ^9.0.0

This means that without warning they then get updates to sub packages like bindings to the very latest version and it becomes difficult for them to troubleshoot and identify which version actually introduced the problem.

## Next Steps

Need to investigate how we can handle this with Lerna.  Unfortunately Lerna makes lots of hints to Babel as a reference implementation, however Babel no longer uses Lerna 🤔 So need to investigate further

reconbot, 2021-06-15:
We should prototype fixed version mode with the carat removed from deps version number in the parent package's package.json on a sample project.

reconbot, 2021-06-19:
#2275 did not work, didn't hurt, but didn't work. When I deployed the latest version of serialport it changed it from a pinned version to a ranged version

GazHank, 2021-06-19:
Do you think this could be a limitation of Lerna, or just more that more config changes are needed?

reconbot, 2021-06-19:
I think it's maybe a limitation of lerna. Might be able to mitigate it with a plug-in? We should open an issue there and ask about it.

I started exploring npm workspaces and yarn workspaces but I have no opinions yet.

Ive also considered breaking up the monorepo so it's easier to use typescript and different release tools (I'd love to bundle the binaries inside the bindings package, got a working poc but I don't know how to get it into lerna)

GazHank, 2021-06-19:
@reconbot looks like we need to use `lerna publish --exact `

per https://github.com/lerna/lerna/issues/2836

If this doesnt work we should probably add to that issue rather than making a new one?

reconbot, 2021-06-19:
Good find!

GazHank, 2021-09-29:
I think this is now resolved :-)  please shout if not

closed 2021-09-29 by GazHank