serialport/node-serialport #2272: binding.read reads only 32 bytes

germancmartinez, 2021-06-18:
### Summary of Problem

When I try to read a large number of bytes (514 in this case) the function ends and returns only 32 bytes. 

### Code 

```js
...
var bytesRead;
var InBuf = Buffer.allocUnsafe(600); 
var portOptions = { baudRate: 9600, dataBits: 8, parity: "even", stopBits: 1, autoOpen: true };  
const binding = new Binding(portOptions);
await binding.open("COM3",portOptions);
...
await binding.write(Buffer.from('1006','hex'));   //OK
bytesRead = await binding.read(InBuf,0,514);  //NOT OK-->I get only 32 bytes

```
I'm using a Serial Port Monitor where I can see:
[image]

### Versions, Operating System and Hardware

- SerialPort v9.1.0
- Node.js v14.17.0
- Operating system and version: Windows Server 2012 R2
- Architecture: x64

reconbot, 2021-06-20:
Hey, sorry the docs are a little light on the bindings. But it will return a minimum of 1 byte and a maximum of specified bytes. You have keep reading until you get the number of bytes you want, read only stays pending while there isn't any data to read.

germancmartinez, 2021-06-21:
Hi Francis, thank you! Using a while is working ok.

myfreax, 2024-03-05:
@germancmartinez  How done fix this issue?

germancmartinez, 2024-03-06:
var portOptions = { baudRate: 9600, dataBits: 8, parity: "even", stopBits: 1, autoOpen: true };   
const binding = new Binding(portOptions);

async function mySerialRead(buff, offset, bytes){
    var bytesRead=0;
    var aux;
    var getOut=false;
    while(bytesRead<bytes) {
        if (getOut) { return bytesRead;}
        aux = await binding.read(buff,offset+bytesRead,bytes-bytesRead);
        bytesRead += aux.bytesRead;
    }
    return bytesRead;
}

myfreax, 2024-03-17:
> ```
> var portOptions = { baudRate: 9600, dataBits: 8, parity: "even", stopBits: 1, autoOpen: true };   
> const binding = new Binding(portOptions);
> 
> async function mySerialRead(buff, offset, bytes){
>     var bytesRead=0;
>     var aux;
>     while(bytesRead<bytes) `{`
>         aux = await binding.read(buff,offset+bytesRead,bytes-bytesRead);
>         bytesRead += aux.bytesRead;
>     }
>     return bytesRead;
> }
> ```

thank your work. I have solved the problem using parser-delimiter package