serialport/node-serialport #2273: Suggestions for improving access devices VID and PID

seqiang365, 2021-06-18:
### Summary of Problem

(Please answer all 3)

- Perhaps because of a hardware driver problem, the path to the device instance under Windows does not contain PID. 
- This problem may be due to the fact that the hardware driver developer did not build the code according to the specification. Is it possible to use the interface of the hardware ID to get VID and PID? 
- And you can set a separate field to hold the VID or PID

### Versions, Operating System and Hardware

- SerialPort@9.0.0
- Node.js v14.17.0
- Electron (if applicable) v13.0.1
- Operating system and version: Windows10
- Architecture: x64
- Hardware and chipset? 5G Model (Other: USB to Serial Port, AT command for receiving and receiving)

GazHank, 2021-06-18:
Hi @seqiang365 

Do you mean "path" doesnt contain the PID?  I think this is normal under windows, and I'd usually expect path to be COM3 or some other mapped com port.

For example with my hardware if I use the [electron-serialport example](https://github.com/serialport/electron-serialport/) within the console I get the following port info:

- locationId: "0007.0000.0000.003.000.000.000.000.000"
- manufacturer: "Microsoft"
- path: "COM3"
- pnpId: "USB\VID_1209&PID_2201&MI_00\7&AB06D17&0&0000"
- productId: "2201"
- serialNumber: "<...>"
- vendorId: "1209"

As such the PID and VID should be accessible irrespective of the path.

a cut down version of the code from the example to generate this would be:

```
const SP = require('serialport')
async function listSerialPorts() {
  await SP.list().then((ports, err) => {
    if(err) {
      //do something with the error
      return
    }
    console.log('ports', ports);
  })
}
```

Does your device not report a productId or vendorId?

seqiang365, 2021-06-19:
Hi @GazHank 

Thank you very much for your reply!

Gets the path to the device instance: USB\VID_2C7C&SUBCLASS_00&PROT_00\7&2D6358F1&0&0004
I think the instance path is pnpId, the PID is taken from the instance path.
The hardware ID of my 5G module is as follows: 
USB\Vid_2C7C&Subclass_00&Prot_00
USB\VID_2C7C&PID_0900&REV_0404&MI_04
USB\VID_2C7C&PID_0900&MI_04

I am trying to modify the SerialPort source code to solve this problem, but it is not working well at present.
I know this problem may be caused by the improper design of the driver developer, but I still hope to use hardware ID to solve the problem of not getting PID.

Thanks again for your reply

GazHank, 2021-06-19:
Hi @seqiang365 

Are you able to share the output you get if you send the port list to the debug console?

Cheers

Gaz

seqiang365, 2021-06-19:
[image]

seqiang365, 2021-06-19:
Hi @GazHank

[image]

GazHank, 2021-06-19:
Thanks @seqiang365 

Would I be correct to assume that this data is consistent with the info in device manager and the registry (e.g. local machine SYSTEM\CurrentControlSet\Enum\USB\ )?  That is to say within all of the usual places in Windows it is only possible to see the vendor ID, but not the product id?

If the product ID is available elsewhere in windows then it might be possible that there is an alternative API that could be used.  However per your initial comment, if the device isn't playing nice and sharing it's product ID at all, then I'm not sure if we can code around that.

Ps I don't suppose you know what the product is should be?  Is it a [known device](http://www.linux-usb.org/usb.ids):

2c7c  Quectel Wireless Solutions Co., Ltd.
	0121  EC21 LTE modem
	0125  EC25 LTE modem
	0191  EG91 LTE modem
	0195  EG95 LTE modem
	0296  BG96 CAT-M1/NB-IoT modem
	0306  EG06/EP06/EM06 LTE-A modem
	0435  AG35 LTE modem

seqiang365, 2021-06-19:
Thanks @GazHank

I know some APIs for getting device information：https://docs.microsoft.com/en-us/windows/win32/devinst/setupapi-h

Can you tell me which API I can use to get the hardware ID?
[image]

GazHank, 2021-06-19:
Oh weird, so it's reporting a PID of 0900....  I guess that first  entry is appearing as the "device instance path", and that is what SetupDiGetDeviceInstanceId is returning.  I'm not sure if we can use Setup.API.h to get more data, or if we would could use [wdm.h](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-iogetdeviceproperty)

Do you know how the device is reported on Linux or Mac?  is it reporting the correct PID?

seqiang365, 2021-06-20:
This device is a new product of Quectel, so PID has never appeared before. I can try to use other APIs to get the PID on Windows platforms, but it may need to be verified on other platforms to make a difference.
VID and PID are currently recognized correctly on Linux platforms. It hasn't been tried on the Mac yet, but I'll let you know the results later.

seqiang365, 2021-06-22:
Hi @GazHank, Thank you very much for your technical support. I have recompiled the source code and added a new API to solve this problem on the basis of the original one. I will close this Issues

closed 2021-06-22 by seqiang365

GazHank, 2021-06-22:
Hi @seqiang365, that's great news.  Is that simply a change in your app, or do you think there is an enhancement we could/should make to the windows code for node-serialport?

seqiang365, 2021-07-02:
Hi @GazHank, I think the problem I encountered was caused by the developer's irregularity. Maybe most people did not encounter such problems, so the title of the Issue was just a suggestion.As for Node-SerialPort, I want it to be compatible with more situations, and getting better and better is what we're working towards.Thank you very much for your technical support. Thank you again