serialport/node-serialport #2280: SerialPort.write(): callback never called

kennyyork, 2021-06-22:
### Summary of Problem

- What are you trying to do?

I'm trying to read/write data over the serial port to a standing desk controller.

- What happens?

When I call write, the completion callback never fires.  If I enable debug logs, I see the following printed out in a tight loop:

```
serialport/bindings/unixWrite Starting write 3 bytes offset 0 bytesToWrite 3 +0ms
serialport/bindings/unixWrite write returned: wrote 0 bytes +0ms
```

- What should have happened?

I would expect at least an error to be called instead of an infinite loop

### Code to Reproduce the Issue

In the code below, "Sent" is never printed.

```js
const SerialPort = require('serialport');

const serialport = new SerialPort(
    '/dev/ttyAMA0', {
        baudRate: 9600,
        dataBits: 8,
        parity: 'none',
        lock: false,
    }
)

serialport.on('open', () => {
    console.log("Open");

    serialport.on('date', (buf) => {
        console.log('Rx ' + buf);
    });

    serialport.write([0, 0, 0], (err, num) => {
        console.log('sent');
    });

    setInterval( () => {
        console.log('poke');
        serialport.write([0xF1, 0xF1, 0x29, 0, 0x29, 0x7E]);
    }, 5000);
});
```

### Versions, Operating System and Hardware

- SerialPort@? "serialport": "^9.1.0"
- Node.js v? v14.17.1
- Prebuild? Build from Source? Built from source
- Operating system and version: Linux (Raspberry Pi)
- Architecture: ARM
- Hardware and chipset? Raspberry Pi

GazHank, 2021-06-22:
Hi @kennyyork have you tried using the [drain method](https://serialport.io/docs/api-stream#serialportdrain) after the write?

kennyyork, 2021-06-22:
> Hi @kennyyork have you tried using the [drain method](https://serialport.io/docs/api-stream#serialportdrain) after the write?

I just tried adding a drain after the initial write and get the same result.

```
serialport.write([0, 0, 0]);
serialport.drain();
```

I'm pretty confident my wiring is correct, but if it were not (say I had my TX/RX wires swapped), would I be seeing an error like this?  It seems odd for the underlying fs.write() to not error but also not write any data

reconbot, 2021-06-22:
It would only do that if it were blocked on the write like the control flow pins was not what the driver wanted or something.

reconbot, 2021-07-05:
I'm going to convert this to our new discussion system until we confirm some sort of bug.

closed 2021-07-05 by reconbot