serialport/node-serialport #2281: Empty list on windows with electron 11 and up or node 14 ?

itschrono, 2021-06-22:
Hello,

I have been using serialport for a while and suddently I always have a empty list (promise does resolve succesfully) and with other app I am detecting 2 ports

(serialport test version 8.0.8)
Minimal example (sp.js) :
```js
const SP = require('serialport');

SP.list().then((ports) => {
  console.log({ports})
  console.log({ports})
  console.log({ports})
  console.log({ports})
  console.log({ports})
  console.log({ports})
  console.log({ports})
  console.log({ports})
  console.log({ports})
}).catch((e) => {
  console.error(e)
});
```

```terminal
> electron sp.js
{ ports: [] }
{ ports: [] }
{ ports: [] }
{ ports: [] }
{ ports: [] }
{ ports: [] }
{ ports: [] }
{ ports: [] }
{ ports: [] }
```

Thanks in advance for the help.

I tried within a console with elevated rights but it is the same... (I am using Windows 10 Pro version 20H2 - 19042.1052)

GazHank, 2021-06-22:
Hi @itschrono 

Do you know which specific prior version was working?  

Also, have you tried updating to the latest version of serialport, since I don't think there are any breaking changes, so your code shouldnt need changes to work with v9.  You would also have the added benefit of electron 11 now included in the prebuilds so no need to rebuild the package :-)

ps - I think something like `let portList = await SerialPort.list();` might be simpler and avoid the multiple console logs

itschrono, 2021-06-23:
Hello, still the same with all version from 8 to 9, with electron 11.4.9 or directly node 14.17.1... :(

itschrono, 2021-06-23:
If you try to add a virtual port with USR-VCOM like below:
[image]

Do you see it listed or not ?
Thanks

itschrono, 2021-06-23:
Indeed it seems this module is only listing "physical" serial ports....

GazHank, 2021-06-23:
Hmm, I hadn't realised you were using a virtual Serial Port, but despite this, I seem to get the port listed:

[image]

At the top you can see the USR-VCOM window, of course I haven't connected it to anything*

At the bottom you can see the port listing.  Prior to adding that virtual port it found 2 ports, then with the virtual port added you can see 4 ports.

To run this test I just used a [slightly moddified version of the electron-serialport example](https://github.com/GazHank/electron-serialport)

* wow based on their [website](https://www.pusr.com/support/downloads/usr-vcom-setup-software.html) they are keen to make sure their lawyers are kept busy with litigation ;-)  So I've not connected to anything, and uninstalled that software straight after running this test :-)

itschrono, 2021-06-24:
Oddly enough it is working now, I don't know what happened... The ports was listed with wmic but this module always returned a blank array...

GazHank, 2021-06-24:
That's good news :-)

If you are able to reproduce the original issue then let me know as I'd be happy to help see if there is an underlying defect that needs to be fixed, or something we need to explain/warn people about in the docs.

closed 2021-07-05 by GazHank