serialport/node-serialport #2282: New Support / Discussion / Github Issue processes

reconbot, 2021-06-23:
## 💥 Proposal

### What feature you'd like to see
I want to separate out bugs, new features and support and maybe start using the [Issue forms](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository#creating-issue-forms) to gather requirements on bug reports.

## Motivation
I'm inspired but a bunch of other open source projects [graphql-ws](https://github.com/enisdenjo/graphql-ws) for example. That have started to use the discussions tab in github. It's super clear what open issues there are with the project and where to go for help.

## Pitch

I want to move proposals and help issues to discussions, and keep issues for accepted proposals and confirmed bugs. I think once that's in place issue forms might help ensure good bug reports.

reconbot, 2021-06-23:
cc @HipsterBrown @GazHank @fivdi @frank-dspeed and probably half a dozen people from support issues that have been incredible at helping others.

GazHank, 2021-06-23:
If it results in zero open issues (per graphql-ws) then who could argue with that!

Just let me know how I can help 👍🏼

frank-dspeed, 2021-06-26:
Splitting it up is clear a hugh win while issues get the most attention i think handling issues more clear is better then splitting. Unless the splitted diskussions get as much attention as issues. So if you got a big team that handels that go for that. If not Stay with issues.

reconbot, 2021-07-05:
It's done https://github.com/serialport/node-serialport/discussions/2285

closed 2021-07-05 by reconbot

reconbot, 2021-11-10:
Added a bug issue form and directed people to discussions for support.