serialport/node-serialport #2284: Parity "even" does not work

jmades, 2021-06-28:
When I change the parity from "none" to "even" the port behaves eactly like parity "none".

Test performed on a Raspberry Pi 3 with RS485 shield as sender and a PC with RS485-2-USB converter on running docklight.

GazHank, 2021-06-28:
Hi @jmades,

are you able to share a little more information about your environment, and perhaps a sample of (or link to) the code that you are trying to use?

- SerialPort v?
- Node.js v?
- Electron (if applicable) v?
- Prebuild? Build from Source?

jmades, 2021-06-28:
> 
> 
> Hi @jmades,
> 
> are you able to share a little more information about your environment, and perhaps a sample of (or link to) the code that you are trying to use?
> 
>     * SerialPort v?
8.0.7
> 
>     * Node.js v?
12.22.1
> 
>     * Electron (if applicable) v?
not used
> 

>     * Prebuild? Build from Source?
Build from source

`import SerialPort = require('serialport');

let message: string = 'Hello World';
console.log(message);
test_connect_config();

function test_connect_config() {

    console.log("Connecting...");

    const port = new SerialPort('/dev/ttyS0', {
            baudRate: 9600,
            dataBits: 8,
            stopBits: 1,
            parity: 'even'
        },
        error => {
            if (error !== null) {
                console.error(error);
            }
        }
    );

    port.write('test', (error?: Error | null) => {});
}
`

GazHank, 2021-06-30:
hi @jmades is it possible that this is related to https://github.com/serialport/node-serialport/issues/1413? If so it looks like the thread to address this may be found at https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=198999

jmades, 2021-06-30:
Thanks, this solved my problem.

closed 2021-06-30 by jmades