serialport/node-serialport #2292: Prebuild for more environments

GazHank, 2021-07-05:
## 💥 Proposal

### What feature you'd like to see
Expand the number of prebuild environments, and increase compatability.  (Perhaps using docker images rather than the GitHub machines directly, giving greater control of the build environment).

Suggested environments:
- [ ]  ARM
- [ ]  Alpine Linux (or other musl based environments)
- [ ]  Apple M1 (Currently blocked by hardware access)

## Motivation
At present it is necessary to build from source for several platforms, and in the case of Linux the environment is automatically upgraded by Github as part of their regular maintennance, therefore compatibility can vary from version to version with little prior notice.  See https://github.com/serialport/node-serialport/issues/2266

## Pitch

Many of the support requests relate to issues with the build process, which should be reduced if prebuilt binaries were available for these platforms.

GazHank, 2021-07-05:
Prebuild for ARM has also been requested before: https://github.com/serialport/node-serialport/issues/2198

Prebuild for Apple M1 previously raised in https://github.com/serialport/node-serialport/issues/2243

GazHank, 2021-07-05:
For ARM, my hypothesis (perhaps flawed) is that we could use a docker image such as those available from [dockcross](https://github.com/dockcross/dockcross) to build for ARM using the Linux x86_64 machine on Github actions.

I'm not sure if this docker based approach is the best solution to improving Linux compatability across distributions, or if we could achieve the same results by just configuring the github actions to use Ubuntu1804 rather than latest?

thegecko, 2021-11-29:
In [node-usb](https://github.com/node-usb/node-usb) we switched to `prebuildify` and `prebuildify-cross` on GH Actions to gain the broader architecture support.

shenzhuxi, 2022-08-25:
Are the prebuilds from bindings-cpp working? https://github.com/serialport/bindings-cpp/blob/main/.github/workflows/build.yml
If so, how can I use them? Currently, I install all the native modules with --build-from-source to run on ARM.

Julusian, 2022-10-08:
In a couple of projects, I have started using a matrix in github actions to build for all desired platforms. Im pretty happy with how it works there https://github.com/Julusian/node-freetype2/blob/master/.github/workflows/node.yaml. 
Similar principle to prebuildify-cross, but I feel like I have more control over the process

thegecko, 2022-10-08:
ooh neat!