serialport/node-serialport #2295: Update packages to remove vulnerabilities introduced by package hoek and cryptiles

paimon0715, 2021-07-06:
Hi,@reconbot
### Issue

   2 vulnerabilities (1 high and 1 medium severity) are introduced in **_serialport_**:
1.Vulnerability **CVE-2018-3728** (medium severity) is detected in package **_hoek_** (versions: <4.2.1,>=5.0.0 <5.0.3):  [https://snyk.io/vuln/npm:hoek:20180212](https://snyk.io/vuln/npm:hoek:20180212)
2.Vulnerability **npmjs-advisories-1464** (high severity) is detected in package **_cryptiles_** (versions: >=0.0.1 <4.1.2):  [https://www.npmjs.com/advisories/1464](https://www.npmjs.com/advisories/1464)
  The above vulnerable packages are transitively referenced by **_serialport_** via:
1.serialport@4.0.7 ➔ node-pre-gyp@0.6.39 ➔ hawk@3.1.3 ➔ boom@2.10.1 ➔ hoek@2.16.3
2.serialport@4.0.7 ➔ node-pre-gyp@0.6.39 ➔ hawk@3.1.3 ➔ cryptiles@2.0.5

### Solution
Since **_serialport@4.0.*_** is transitively referenced by **221** downstream projects (e.g., [@code-dot-org/johnny-five 0.11.1-cdo.2](https://www.npmjs.com/package/@code-dot-org/johnny-five) (latest version),[discobus 1.0.2](https://www.npmjs.com/package/discobus) (latest version), [octalbonescript 1.3.1](https://www.npmjs.com/package/octalbonescript)  (latest version), [thing-it-device-microcontroller 0.8.3](https://www.npmjs.com/package/thing-it-device-microcontroller) (latest version), [filament-sty 2.0.18](https://www.npmjs.com/package/filament-sty)(latest version)), 

**_serialport@1.7.*_** is referenced by **102** downstream projects (e.g., [grbljs 0.1.0](https://www.npmjs.com/package/grbljs) (latest version), [spark-cli 1.4.2](https://www.npmjs.com/package/spark-cli) (latest version), [iot-client 2.8.12](https://www.npmjs.com/package/iot-client) (latest version), [pimatic-jeelabs 0.8.7](https://www.npmjs.com/package/pimatic-jeelabs) (latest version), [pimatic-can 0.1.3](https://www.npmjs.com/package/pimatic-can) (latest version)),

**_serialport@2.1.*_** is referenced by **67** downstream projects (e.g., [aquila-server 0.2.3](https://www.npmjs.com/package/aquila-server) (latest version), [node-red-habanero 1.1.20](https://www.npmjs.com/package/node-red-habanero) (latest version), [pinoccio 0.1.40](https://www.npmjs.com/package/pinoccio) (latest version), [coap-shepherd 0.3.1](https://www.npmjs.com/package/coap-shepherd) (latest version), [cylon-sphero 0.23.0](https://www.npmjs.com/package/cylon-sphero) (latest version)),

If **_serialport_** removes the vulnerable packages from the above versions, then its fixed versions can help downstream users decrease their pain.

Could you help update packages in these versions?

### Fixing suggestions

(1）In **_serialport@4.0.*_**, you can kindly perform the following upgrades (not crossing their major versions):
`node-pre-gyp ^0.6.32 ➔^0.7.0`;  

**Note:**
_node-pre-gyp@0.7.0,>0.7.0 transitively depends on hoek@4.2.1(a vulnerability CVE-2018-3728 patched version)_

(2）In **_serialport@1.7.*_**, you can kindly perform the following upgrades (not crossing their major versions):
`node-pre-gyp 0.6.x ➔^0.7.0`;  

**Note:**
_node-pre-gyp@0.7.0,>0.7.0 transitively depends on hoek@4.2.1(a vulnerability CVE-2018-3728 patched version)_

(3）In **_serialport@2.1.*_**, you can kindly perform the following upgrades (not crossing their major versions):
`node-pre-gyp ^0.6.26 ➔^0.7.0`;  

**Note:**
_node-pre-gyp@0.7.0,>0.7.0 transitively depends on hoek@4.2.1(a vulnerability CVE-2018-3728 patched version)_

Thanks for your contributions to the npm ecosystem!

Best regards,
Paimon

GazHank, 2021-07-06:
Sorry, but I think people will need to be encouraged to upgrade to a more recent version of Serialport as these issues are already addressed there.  If someone hasn't updated their package dependencies since 2017 (when version 5 was released) then I suspect there will be a number of security and support issues they will face.

The problem is that version 4 of serial port works with node version 6 and prior which are long out of support.  I don't think we could even build them in the GitHub actions environment without issues.

If there are actively used packages where people would like some support to migrate to a newer version of Serialport then I'd be happy to try to help with that migration (I think this would be far safer than for the package to stick with v4)

paimon0715, 2021-07-06:
@GazHank Thanks for your feedback.
A lot of downstream users transitively use serialport@1.7.* and @2.1.* via unmaintained packages (the unmaintained packages cannot upgrade serialport from 1.7.* or 2.1.* to version 4). This is the root cause.

So if you can kindly release a fixed version, the downstream users can avoid introducing vulnerablities.

GazHank, 2021-07-06:
If there is no appitite to update the unmaintained packages, but people still want to use them without upgrading then the alternative option might be to use [Selective dependency resolutions](https://classic.yarnpkg.com/en/docs/selective-version-resolutions/)

I haven't had chance to properly look at the vulnerabilities, but since they relate to the prebuild tooling, rather than the released code, I'm not sure if the vulnerability can actually 

Since we use Github hosted runners, we are limited to a [small set of build environment options](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-software), so arent able to build those old versions any more.  Additionally if we were to publish an old version without the associated prebuild binaries then it is likely to cause even more issues as the install process for the package will fail when it can't find the tar ball.  (and I think most people wont know how to build tthat old code from source)

Of course if you are able to pull any string with the git environments, or help setup some containers or similar to be able to target those old environments then that help would be greatly appreciated :-)

Cheers

Gaz

reconbot, 2021-07-06:
I'm happy to depreciate those versions but I'm not updating years old versions of serialport because of vulns in dev deps. They don't even work in Lts versions of nodejs.

reconbot, 2021-07-06:
I've depreciated the last versions from each of those major versions and a few others that have had a few downloads in the last few months.

We have a [security policy for disclosure](https://github.com/serialport/node-serialport/blob/master/SECURITY.md) if this was something concerning I'd appreciate you not reporting it publicly until we have some time to address it.

closed 2021-07-06 by reconbot

GazHank, 2021-07-06:
@reconbot sorry for triggering the MSFT bot, I guess it was because I proposed to use dependabot, and ran it against my fork of the repo :-(