serialport/node-serialport #2296: Installation Error

yy2428, 2021-07-11:
## 👉 [Please follow one of these issue templates](https://github.com/serialport/node-serialport/issues/new/choose) 👈

Hello,

I tried to install serialport package, but ran into this error:
PS E:\tmp\serialport> npm install serialport
npm ERR! code 1
npm ERR! path E:\tmp\serialport\node_modules\@serialport\bindings
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@16.4.2 | win32 | x64
npm ERR! gyp info find Python using Python version 3.8.6 found at "D:\apps\msys2\mingw64\bin\python3.exe"
npm ERR! gyp http GET https://nodejs.org/dist/v0.52.2/node-v0.52.2.tar.gz
npm ERR! gyp http 404 https://nodejs.org/dist/v0.52.2/node-v0.52.2.tar.gz
npm ERR! gyp WARN install got an error, rolling back install
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack Error: 404 response downloading https://nodejs.org/dist/v0.52.2/node-v0.52.2.tar.gz
npm ERR! gyp ERR! stack     at Request.<anonymous> (D:\apps\nodejs\node_modules\npm\node_modules\node-gyp\lib\install.js:187:14)
npm ERR! gyp ERR! stack     at Request.emit (node:events:406:35)
npm ERR! gyp ERR! stack     at Request.onRequestResponse (D:\apps\nodejs\node_modules\npm\node_modules\request\request.js:1059:10)
npm ERR! gyp ERR! stack     at ClientRequest.emit (node:events:394:28)
npm ERR! gyp ERR! stack     at HTTPParser.parserOnIncomingClient (node:_http_client:621:27)
npm ERR! gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (node:_http_common:127:17)
npm ERR! gyp ERR! stack     at TLSSocket.socketOnData (node:_http_client:487:22)
npm ERR! gyp ERR! stack     at TLSSocket.emit (node:events:394:28)
npm ERR! gyp ERR! stack     at addChunk (node:internal/streams/readable:312:12)
npm ERR! gyp ERR! stack     at readableAddChunk (node:internal/streams/readable:287:9)
npm ERR! gyp ERR! System Windows_NT 10.0.19041
npm ERR! gyp ERR! command "D:\\apps\\nodejs\\node.exe" "D:\\apps\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd E:\tmp\serialport\node_modules\@serialport\bindings
npm ERR! gyp ERR! node -v v16.4.2
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok

It seems the issue is to ask node-gyp to target and get node-v0.52.2.tar.gz that no longer exists.  Please advice how to get around this error.  Thanks.  

Yong

GazHank, 2021-07-11:
Hi @yy2428 would you be able to confirm a little more info about your environment please?

It looks like you are using windows and trying to build from source (rather than using the prebuilt binaries)?

It would be useful if you can confirm the versions you are using:
- Serialport
- Node
- Electron (if applicable)

yy2428, 2021-07-11:
Hi Gareth, thank you for your quick reply.  I am using Windows 10, node -v v16.4.2, node-gyp -v v7.1.2.  Since I  am  using “npm install serialport”, the Serialport version should be the current 9.2.0 on npmjs.com.

It was mentioned on your website that there is a prebuilt binaries, but I could not find it.  Can you send me the  link for me to try it out?  Thank you.  Yong

GazHank, 2021-07-11:
Thanks @yy2428 if it hasn't been overridden by config rules then the installer will default to downloading the prebuild if a version exists for you environment (which they should given the windows and node versions you are using).

Is this an existing project that you are adding serialport to, or a new (/blank) project? If it is an existing project are their any config rule or scripts in your project.Json which would affect the install process?

Also is there any chance that there are any issues with your node environment or npm config settings?

yy2428, 2021-07-11:
This is a new(blank) project, with newly installed node and npm just for trying out node-serialport.  node and npm seem to work fine.  I guess the issue is that https://nodejs.org/dist/v0.52.2/node-v0.52.2.tar.gz  maybe hard coded somewhere in the serialport package?

GazHank, 2021-07-11:
Thanks I'll try to see if I can find what is going on.  It does look like an environment issue to be, but I will see if I can find what's going on.

If you are trying the package out for the first time then you could try the example at https://github.com/serialport/electron-serialport#to-use While that is an electron example, it does make it quite easy to get a sense of how the package works.  Of course that project would also need to perform an install, so it might allow you to check if you get the same errors

yy2428, 2021-07-11:
Just tried it, electron-serialport, same error cam up: 
http 404 https://nodejs.org/dist/v0.52.2/node-v0.52.2.tar.gz

Why is node-v0.52.2.tar.gz needed?  Can you change it to node-v16.x?

GazHank, 2021-07-11:
node-v0.52.2.tar.gz isn't needed, it looks like the node install is not working fully.

how did you install node, did you use a package manager?

yy2428, 2021-07-11:
I just downloaded node-v16.4.2-x64.msi and installed it according to its instructions.  It was a smooth run.  npm came with it naturally.

GazHank, 2021-07-11:
In the first instance I'm not sure why your machine isn't just downloading the prebuit binaries from https://github.com/serialport/node-serialport/releases/tag/@serialport/bindings@9.2.0 since bindings-v9.2.0-node-v93-win32-x64.tar.gz would be the right version for your machine...

I assume you've not had node installed on that machine before?  I've been searching for others with similar issues, and the only common thread is where people have conflicting versions, or misconfigured environment