serialport/node-serialport #2299: Errors installing electron-serialport example on windows 10

gunnarSensus, 2021-07-19:
i followed the instructions to install and run the electron-serialport example project, but am encountering the following error when installing on windows 10, node: v14.17.3, npm: 6.14.13

```
PS C:\Users\rannug\Documents\work\electron-serialport> npm install
npm WARN read-shrinkwrap This version of npm is compatible with lockfileVersion@1, but package-lock.json was generated for lockfileVersion@2. I'll try to do my best with it!
(node:1280) [DEP0123] DeprecationWarning: Setting the TLS ServerName to an IP address is not permitted by RFC 6066. This will be ignored in a future version.
(Use `node --trace-deprecation ...` to show where the warning was created)
npm WARN deprecated node-pre-gyp@0.11.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.

> lzma-native@6.0.1 install C:\Users\rannug\Documents\work\electron-serialport\node_modules\lzma-native
> node-pre-gyp install --fallback-to-build && rimraf build

node-pre-gyp WARN Using request for node-pre-gyp https download
[lzma-native] Success: "C:\Users\rannug\Documents\work\electron-serialport\node_modules\lzma-native\binding-v6.0.1-n-api-win32-x64\lzma_native.node" is installed via remote

> @serialport/bindings@9.2.0 install C:\Users\rannug\Documents\work\electron-serialport\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> core-js@3.15.2 postinstall C:\Users\rannug\Documents\work\electron-serialport\node_modules\core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> electron@13.1.7 postinstall C:\Users\rannug\Documents\work\electron-serialport\node_modules\electron
> node install.js

> electron-serialport@1.0.2 install C:\Users\rannug\Documents\work\electron-serialport
> electron-rebuild

⠦ Building module: bindings, Completed: 0Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
⠙ Building module: bindings, Completed: 0  serialport.cpp
⠦ Building module: bindings, Completed: 0c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(1652): error C2059: syntax error: '' [C:\Users\rannug\Documents\
work\electron-serialport\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(1654): error C2238: unexpected token(s) preceding ';' [C:\Users\
rannug\Documents\work\electron-serialport\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(1670): error C2061: syntax error: identifier 'ResolveCallback' [
C:\Users\rannug\Documents\work\electron-serialport\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(7292): error C2059: syntax error: '' [C:\Users\rannug\Documents\
work\electron-serialport\node_modules\@serialport\bindings\build\bindings.vcxproj]
c:\users\rannug\.electron-gyp\13.1.7\include\node\v8.h(8652): error C2061: syntax error: identifier 'HostImportModuleDy
namicallyCallback' [C:\Users\rannug\Documents\work\electron-serialport\node_modules\@serialport\bindings\build\bindings
.vcxproj]
✖ Rebuild Failed

node-gyp failed to rebuild 'C:\Users\rannug\Documents\work\electron-serialport\node_modules\@serialport\bindings'.

Error: node-gyp failed to rebuild 'C:\Users\rannug\Documents\work\electron-serialport\node_modules\@serialport\bindings'.
Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1

    at ModuleRebuilder.rebuildNodeGypModule (C:\Users\rannug\Documents\work\electron-serialport\node_modules\electron-rebuild\lib\src\module-rebuilder.js:193:19)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at async Rebuilder.rebuildModuleAt (C:\Users\rannug\Documents\work\electron-serialport\node_modules\electron-rebuild\lib\src\rebuild.js:190:9)
    at async Rebuilder.rebuild (C:\Users\rannug\Documents\work\electron-serialport\node_modules\electron-rebuild\lib\src\rebuild.js:152:17)
    at async C:\Users\rannug\Documents\work\electron-serialport\node_modules\electron-rebuild\lib\src\cli.js:146:9
npm ERR! code ELIFECYCLE
npm ERR! errno 4294967295
npm ERR! electron-serialport@1.0.2 install: `electron-rebuild`
npm ERR! Exit status 4294967295
npm ERR!
npm ERR! Failed at the electron-serialport@1.0.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\rannug\AppData\Roaming\npm-cache\_logs\2021-07-19T19_31_09_998Z-debug.log
PS C:\Users\rannug\Documents\work\electron-serialport> node -v
v14.17.3
PS C:\Users\rannug\Documents\work\electron-serialport> npm -v
6.14.13
PS C:\Users\rannug\Documents\work\electron-serialport>```

![image](https://user-images.githubusercontent.com/13751910/126216851-fb18c086-c09d-478b-850b-a205c7ba7a1b.png)

Any help to resolve the error would be appreciated, Thank you.

GazHank, 2021-07-19:
Hi @gunnarSensus 

I noticed the "C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe" reference (which is based on VS 2015), and wonder if you might need to update the [build tooling to allow node-gyp to work](https://github.com/nodejs/node-gyp#on-windows)

I'm runing the VS 2019 build tools, but think that the 2017 tools should work too

gunnarSensus, 2021-07-20:
I followed the instructions in the node-gyp link and it is happy now :) 
[image]

Thank you so much for helping me figure it out, really appreciate it! just wish native modules werent so complicated in node :)

closed 2021-07-20 by gunnarSensus

GazHank, 2021-07-20:
I completely agree

jacobq, 2021-07-26:
FWIW I ran into this after removing VS build tools 2015/2017 in favor of 2019 (for [Tauri setup](https://tauri.studio/en/docs/getting-started/setup-windows/)). Reinstalling 2017 fixed the problem.

GazHank, 2021-07-26:
@jacobq did you have the build location included in your environment path for VS 2019?

I think if you install the build tools via npm (rather than a MSFT download), the installation process still sets the environment variables correctly.  However I think that version is still based on 2017, so switching to 2019 requires a little manual tinkering for now.

jacobq, 2021-07-26:
> @jacobq did you have the build location included in your environment path for VS 2019?

Sorry, I did not check or try to troubleshoot/diagnose further.