serialport/node-serialport #2300: Raspbetty. Dividing one UART an answer on a few parts, on a new Linux kernel

GerasimGerasimov, 2021-07-30:
## 📚 Documentation

**Dividing one UART an answer on a few parts, on a new Linux kernel**

Not matter Node.JS 10 or 14.
I've tryed on Raspberry: CM3+, Pi 4 Model B, Pi 3 Model B

**Hardware connection:**
RPi (Node serialport) <->USB/RS485<->device

**That i'm doing:**
I'm requesting one time a device  for a binary data (Modbus RTU)  by uart, and a device send me one response. Quantity of bytes in response, not known in advance. End of response ("data" event) emitted when a 1.5 bytes-lenght time-delay after a last byte has expired. And that approach have been working long time on old Linux 5.4.51(as well as it still excellent working on x86 Windows 7, 10, Ubuntu Linux)

**Normar behavior:** i have received one whole an answer from a device. 

**Incorrect behavior:** An one device's respond has been divided at a few parts, as result i have received a few data packets. 

**Normal behavior**: Linux 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux

**Incorrect behavior**: Linux 5.10.44-v7+ #1427 SMP Wed Jun 23 13:49:47 BST 2021 armv7l GNU/Linux

What was be changed at a new RPi Linux kernel? And how to reliably do my case without a sifting through the operating systems.

At Windows API i can set delay after data packet:
    commtimeouts.ReadIntervalTimeout         = after_packet_delay;
    commtimeouts.ReadTotalTimeoutConstant    = 500;//device not respond
   ...
    SetCommTimeouts( hCommFile, commtimeouts);

What about Linux? Are you setting commtimeouts by default?

GerasimGerasimov, 2021-08-04:
i have asked about at raspberry's forum
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=317111
UART behavior depends on type of device /dev/ttyXXX

GazHank, 2021-08-04:
Hi @GerasimGerasimov 

In case this helps, the read / write functionality are a little different between Windows and the Unix/POSIX based environments; windows uses the windows fileapi.h, while other platforms use the fs.h.  This would explain differences between linux and windows that you are seeing, but not differences between linux on different architectures.

You could try to use the [InterByteTimeout Parser](https://serialport.io/docs/api-parser-inter-byte-timeout#example) if you want to set timeouts, but I'm not sure if that would be helpful for your usage

GerasimGerasimov, 2021-08-15:
Thanks, i have used the idea from your advice. The 'data' event is gathering chunks of one respond and starting a timer to frame-end delay. And once a timer emitted, the whole respond is ready to processing.

```typescript
ChunkEndTime: number = 10;
Respond: Array<number> = [];

port.on('data',  (data: any)=>{
  Respond.push(...data);
  restartChunkEndTimer(ChunkEndTime)
});

function restartChunkEndTimer(frameEndTime: number):void {
  setTimeout(() => {
    onChunkEndTime();
    }, frameEndTime);
 }

function onChunkEndTime():void {
  console.log(`respond: ${Respond}`);
  processingRespond(Respond);
 }

function processingRespond(resp: Array<number>) {
.....
}
```

closed 2021-08-15 by GerasimGerasimov