serialport/node-serialport #2301: `GLIBC_2.28' not found

joaorbdg, 2021-08-01:
Hello, I'm trying to run an electron application on Ubuntu 20.04 using the @serialport package, and the app is being installed through the Snapcraft store.

But before the app can start, is throwing the following exception related to the @serialport

```
#033[1m#033[47m#033[31mA JavaScript error occurred in the main process
#033[30mUncaught Exception:
Error: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /run/user/1000/snap.xxx/.org.chromium.Chromium.juyTIN)
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5:1846)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1138:18)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5:2073)
    at Module.load (internal/modules/cjs/loader.js:935:32)
    at Module._load (internal/modules/cjs/loader.js:776:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
    at Module.require (internal/modules/cjs/loader.js:959:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at bindings (/snap/xxx/9/resources/app.asar/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/snap/xxx/9/resources/app.asar/node_modules/@serialport/bindings/lib/linux.js:2:36)#033[0m
```

Does anybody have some idea what can be happening?

GazHank, 2021-08-01:
Hi @joaorbdg which app is it that you are using from snapcraft store?

joaorbdg, 2021-08-01:
@GazHank It is a private app. Btw, I'm building it using `electron-builder`

GazHank, 2021-08-01:
What version of glibc do you have installed?

Also, are you packaging the app, and installing on the same machine?  To me it looks like the machine used to build the package/app is running newer versions of the build tools than the machine you are trying to install to.

If this is the case then it is advisable to build using the oldest versions of the libraries that you would like to support

joaorbdg, 2021-08-01:
I'm building/packaging the app on Ubuntu 20.04 which has the GLIBC 2.31

Output from `ldd --version`
```
ldd (Ubuntu GLIBC 2.31-0ubuntu9.2) 2.31
```

So, based on your thought, I'll try to build/package the app on Ubuntu 18.04.5, which contains the following GLIBC
```
ldd (Ubuntu GLIBC 2.27-3ubuntu1.4) 2.27
```

joaorbdg, 2021-08-01:
I've packaged the application on Ubuntu 18.04.5, but the same issue persists. If I run the application locally, works fine. This issue is just happening when the application is packaged to distribute on the Snapcraft store.

joaorbdg, 2021-08-01:
I've deployed a blank app based on this repo https://github.com/serialport/electron-serialport and the same issue occurs:

[image]

You can install the app [Electron Serial Port](https://snapcraft.io/blank-demo-app) on Ubuntu running:
```
snap install blank-demo-app
```

I've used Ubuntu 18.04.5 to package the application.

GazHank, 2021-08-01:
Thanks @joaorbdg 

Based on the fact that the app runs correctly until packaged, it would seem to be a problem with electron-builder or it's configuration.

I assume you are using `"buildDependenciesFromSource": true,"npmRebuild": false,` in your project's json, rather than the prebuild binaries?

I don't have access to a linux machine at the moment, but will try to take a closer look when I get access, if you are still facing issues

joaom182, 2021-08-01:
@GazHank Good catch! I'll try to add these configs to my package.json and check again.

joaorbdg, 2021-08-02:
@GazHank it worked! the issue was because I didn't add `buildDependenciesFromSource` and `npmRebuild` to the `electron-builder.yml` file.

Thank you so much!

closed 2021-08-02 by joaorbdg

Mumbaso21, 2024-05-28:
For all those who are still having problems with this topic, I report the configuration that worked for me.

Environment:
```
Windows 10 Home 64bit 22H2 (19045)
WSL 2 for windows enabled (Hyper-V is not available on Windows 10 Home)
Docker Desktop
electron: 13.0.1
electron-builder: 23.0.3 (used with Docker)
serialport: 10.4.0
@serialport/parser-readline: 10.3.0
```

Objective:
```
create a build from Windows to Ubuntu 18.04
```

As you can read at this [link](https://www.electron.build/multi-platform-build#docker), `electron-builder` allows the use of `docker` to create builds from Windows to Linux-like, in particular thanks to the images made available directly by them you can choose which image to use to create the build with.
I tested both the `electronuserland/builder` version (with NodeJs 18) and the `electronuserland/builder:16` version (with NodeJs 16) and both worked after correctly configuring the package.json file in electron.
The images contain Ubuntu 20.04 and NodeJs 18 or 16 depending on the version you choose.

To make the builds work correctly I set the parameters suggested above by @GazHank like this:
```
"buildDependenciesFromSource": false,
"npmRebuild": false
```

Previously I tried:
```
"buildDependenciesFromSource": true,
"npmRebuild": false
```

but it didn't work, it gave an error `GLIBC_2.28 not found`.

I hope it helps