serialport/node-serialport #2303: Installation issue

DevinKott-KA, 2021-08-03:
I built an application using the `serialport@8.0.8` package was able to call `let list = await SerialPortStream.list()`. When I switched to the precompiled bindings (available from the releases page), I am now getting `UnhandledPromiseRejectionWarning: TypeError: SerialPortStream.list is not a function`.

How can I get this function back? I printed out the bindings `require` and got the following functions:

- set
- get
- getBaudRate
- open
- update
- close
- flush
- drain
- Poller

Am I using the wrong bindings file, or will I have to rewrite what I have to use only those functions?

Thanks.

GazHank, 2021-08-03:
Hi @DevinKott-KA, are you able to share a view of what changed within your package.Json between the version that works to the version which fails?  (I think this would help understand what changed and what broke)

Also, are there significant issues that are blocking you from upgrading to a more recent version on node?

DevinKott-KA, 2021-08-04:
`package.json` before:

```
{
  "name": "launcher",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1",
    "start": "node index.js"
  },
  "author": "Devin XXXX <XXXXXXXX@gmail.com>",
  "license": "ISC",
  "dependencies": {
    "serialport": "8.0.8"
  }
}
```

`package.json` after:

```
{
  "name": "launcher",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1",
    "start": "node index.js"
  },
  "author": "Devin XXXX <XXXXXXXX@gmail.com>",
  "license": "ISC",
  "dependencies": {}
}

```

And yes, I am blocked to using that version of node since the target machines I'm developing for use that version (which I cannot change).

GazHank, 2021-08-04:
That's strange, you should still have serialport listed in your dependencies.  

When you install serialport (using yarn or npm), it should automatically choose to use the prebuild binaries if they exist for your machine.  If a version for your combination of node and OS/architecture doesnt exist then it will try to build the package from source, but by default it should use the prebuild binaries, so you shouldn't need to download them manually.

DevinKott-KA, 2021-08-04:
@GazHank  Oh okay, gotcha. I removed the listing from my `package.json` because I thought the bindings file contained everything (my mistake).

If I add it back in, will the entry still be this?

```
"serialport": "^8.0.8"
```

How do I point it to use the bindings file and not build from source, and where should the bindings file be put (folder structure, etc)?

Sorry, this is my first time using a pre-compiled bindings file for node, so I'm not sure how to set everything up.

GazHank, 2021-08-04:
yes that's right for the dependencies.  As part of the [install](https://docs.npmjs.com/cli/v7/commands/npm-install), you can specify the version to not exceed v8 so as to continue to work with your current node version.

By default it should used the pre-compiled binaries since you are on node 8.11.4 it will use Node module version 57.  for which there are builds for mac (64 bit), while for linux and windows both x86 and 64bit are available.

[image]

you shouldn't need to do anything to use the prebuilds.  The system should try to use these first.  However if it doesnt find a suitable version, for example if you were using an ARM processor, then it would try to build from source.

There are options to force a rebuild from source, but from the info you have shared it sounds like you want to use the prebuilt binaries...

DevinKott-KA, 2021-08-06:
Where should I place the bindings file?

If I add the `serialport` dependency, it still tries to build it from source.

I tried placing `bindings.node` in a `build` folder, but no luck. I also tried using the bindings package and importing it as shown:
```
const binds = require('bindings')('bindings.node');
```

But that prints the same set of functions (without `list`) and `npm i` is still trying to build from source.

GazHank, 2021-08-06:
What is your operating system and processor architecture?

DevinKott-KA, 2021-08-06:
@GazHank I am developing on Fedora 34 (Workstation Edition) and my CPU is Intel(R) Core(TM) i7-8700K (x86_64).

GazHank, 2021-08-06:
Adding serialport as a dependency really shouldn't cause a build from source, for a prebuilt environment, unless your npm config is set to build from source.

I don't think we should just add the node file in manually.  I think it would be better to fix your issue with the prebuild or let the system build from source (on a WS install you probably have all of the build tools you need anyway).  Only downside to a build from source is the need to setup the build tools, and the slightly longer install time (seconds)

DevinKott-KA, 2021-08-06:
@GazHank I guess I'm very new to this topic. To create this project on my machine, I ran `npm init`, ran `npm i serialport@8.0.8` (which can build from source fine on my computer), and then made my project.

Now, when placing this project on a target machine, it cannot build `serialport` from source. We tried installing the correct toolings, but could not get everything to work. That is why I'm looking at the pre-compiled binaries so we can skip that entire building part.

I found the precompiled node bindings (on the Releases page) but I'm unsure of how to set up my project for using them instead of building from source. I have no `npm config` and am unsure if I have a prebuild system in place.

GazHank, 2021-08-06:
And the target machine is running the same os and processor architecture?

DevinKott-KA, 2021-08-06:
@GazHank Build machine is Win7, target machine is Win10. Both `x86_64`.

GazHank, 2021-08-06:
Well the bindings.node would sit under node_modules\@serialport\bindings\build\Release\

per below:
[image]

I guess if manually placing the file in this folder works for you then worth a try?

I'm also trying to get my head around what is happening (sorry it's probably just been a long day for me)

So Development is being performed on Linux machine (Fed 34, x86_64) - which is building from source
Build is being performed on Windows 7 x86_64 - also building from source?
Target machine is Windows 10 x86_64 - which tries to build from source but fails

Are all 3 machines running the same version of node?

DevinKott-KA, 2021-08-06:
@GazHank No problem at all, just happy that you're trying to help!

1. Development is on my own linux fedora machine, building from source
2. Build machine builds the programs (runs `npm ci`, etc)
3. Built programs are transferred onto target machines

And yes, I have matched my own version of node to the version running on the build/target machine. I'll give that folder location a shot.

GazHank, 2021-08-06:
Let me know how you get on.  Feel free to share any error logs or other information if you hit any issues.

closed 2021-09-08 by DevinKott-KA