serialport/node-serialport #2304: Build fails when rebuilding for Electron on Windows

ryanplusplus, 2021-08-05:
Sorry for not using a template. I didn't see one that was clearly intended for reporting issues.

I'm rebuilding all native modules for Electron on several platforms. This works for MacOS on ARM, MacOS on AMD64, and Linux on AMD64, but fails on Windows.

Versions:
- node: 14.17.4
- serialport: 9.2.0
- electron: 13.1.8
- electron-builder: 22.11.7

Below is the output that I receive during the rebuild of `serialport`. Note that I've configured `electron-builder` to rebuild from source instead of pre-builts because we encounter a runtime error using `serialport` on Linux AMD64 that, I think, is because of a bad or incompatible prebuilt.

```
    C:\a\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\nvm\v14.17.4\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\nvm\v14.17.4\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 
    Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
      serialport.cpp
      serialport_win.cpp
      win_delay_load_hook.cc
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(1652): error C2059: syntax error: '' (compiling source file ..\src\serialport_win.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(1654): error C2238: unexpected token(s) preceding ';' (compiling source file ..\src\serialport_win.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(1670): error C2061: syntax error: identifier 'ResolveCallback' (compiling source file ..\src\serialport_win.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(1652): error C2059: syntax error: '' (compiling source file ..\src\serialport.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(1654): error C2238: unexpected token(s) preceding ';' (compiling source file ..\src\serialport.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(1670): error C2061: syntax error: identifier 'ResolveCallback' (compiling source file ..\src\serialport.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(7292): error C2059: syntax error: '' (compiling source file ..\src\serialport_win.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(8652): error C2061: syntax error: identifier 'HostImportModuleDynamicallyCallback' (compiling source file ..\src\serialport_win.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(7292): error C2059: syntax error: '' (compiling source file ..\src\serialport.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]
    c:\users\slave\.electron-gyp\13.1.8\include\node\v8.h(8652): error C2061: syntax error: identifier 'HostImportModuleDynamicallyCallback' (compiling source file ..\src\serialport.cpp) [C:\a\node_modules\@serialport\bindings\build\bindings.vcxproj]

    
                    errorOut=gyp ERR! build error 
    gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1
    gyp ERR! stack     at ChildProcess.onExit (C:\nvm\v14.17.4\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
    gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
    gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
    gyp ERR! System Windows_NT 10.0.19043
    gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\nvm\\v14.17.4\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
    gyp ERR! cwd C:\a\node_modules\@serialport\bindings
    gyp ERR! node -v v14.17.4
    gyp ERR! node-gyp -v v5.1.0
    gyp ERR! not ok 
    npm ERR! code ELIFECYCLE
    npm ERR! errno 1
    npm ERR! @serialport/bindings@9.2.0 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
    npm ERR! Exit status 1
    npm ERR! 
    npm ERR! Failed at the @serialport/bindings@9.2.0 install script.
    npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
    
    npm ERR! A complete log of this run can be found in:
    npm ERR!     C:\Users\jenkins\AppData\Roaming\npm-cache\_logs\2021-08-04T22_41_33_386Z-debug.log
    
                    command='C:\Program Files\nodejs\node.exe' 'C:\nvm\v14.17.4\node_modules\npm\bin\npm-cli.js' rebuild @serialport/bindings@9.2.0 bigint-buffer@1.1.5 node-hid@2.1.1
                    workingDir=
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! ApplCore@1.3.0 package: `electron-builder --publish never`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ApplCore@1.3.0 package script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\jenkins\AppData\Roaming\npm-cache\_logs\2021-08-04T22_41_33_469Z-debug.log
```

And below is the output when `electron-builder` is _not_ instructed to build from source.

```
    > @serialport/bindings@9.2.0 install C:\a\node_modules\@serialport\bindings
[... 48 lines cut ...]
    npm ERR! errno 1
    npm ERR! @serialport/bindings@9.2.0 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
    npm ERR! Exit status 1
    npm ERR! 
    npm ERR! Failed at the @serialport/bindings@9.2.0 install script.
    npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
    
    npm ERR! A complete log of this run can be found in:
    npm ERR!     C:\Users\slave\AppData\Roaming\npm-cache\_logs\2021-08-05T14_09_17_841Z-debug.log
    
                    command='C:\Program Files\nodejs\node.exe' 'C:\nvm\v14.17.4\node_modules\npm\bin\npm-cli.js' rebuild @serialport/bindings@9.2.0 bigint-buffer@1.1.5 node-hid@2.1.1
                    workingDir=
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! ApplCore@1.3.0 package: `electron-builder --publish never`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ApplCore@1.3.0 package script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\slave\AppData\Roaming\npm-cache\_logs\2021-08-05T14_09_17_950Z-debug.log
```

Any help or further direction for debugging this would be greatly appreciated.

reconbot, 2021-08-05:
(I've been encouraging people to use discussions for support as noted in the pinned issue.)

We don't have win32 prebuilt binaries and the system your installing in isn't setup to compile any binaries. Check out our install docs on https://serialport.io to see how to install them.

ryanplusplus, 2021-08-05:
Thanks @reconbot. We've been building with a previous version of `serialport` and `electron` for years on this system so I assumed we were still good to go. I will check out the latest docs.

ryanplusplus, 2021-08-05:
Updating the MS build tools to 2019 (we were on 2015) fixed this issue

closed 2021-08-05 by ryanplusplus