serialport/node-serialport #2309: Writing message async

PatrickHallek, 2021-08-24:
## 💥 Proposal

### What feature you'd like to see
I would find it very helpful to have a async method for writing to the serial port.

## Motivation
I need to make sure that writing a message has been completed. Since the writing of specific messages is part of a pipeline in my use case, it needs to be awaited. I implemented a workaround, but I am not sure if it is a good idea to use the write method like this. A native async approach would be much more clearer and handy. 

## Pitch

```ts
const writeResult = await serialPort
    .writeAsync(message)
    .catch((err) => console.log('Something went wrong', err))
```

## Current workaround
```ts
const writeResult = await new Promise((resolve, reject) => {
    serialPort.write(message, (error) => {
        if (error) return reject(error)
        return resolve()
    })
}).catch((err) => console.log('Something went wrong', err))
```

reconbot, 2021-08-24:
You're using the stream interface which doesn't have async functions. The bindings interface does. I've also always wanted an async iterator interface.

PatrickHallek, 2021-08-24:
Thanks for your quick response!
Unfortunately there is no @types package for the stream interface yet.
Are there any downsides of using my current workaround?

reconbot, 2021-08-24:
`serialport` is the stream interface. I'm unsure if the `@serialport/bindings` package has types. Your workaround should work great!

closed 2021-10-31 by reconbot