serialport/node-serialport #2313: Precompiled binary bindings does not work in electron v14.0.0

udik-chudik, 2021-09-06:
Hi!
After upgrading electron to version 14.0.0 in my project, electron-builder starts to break my builds. It took me about 3 hours to figure out what is going on, so I decide to write this issue to save someone else time:

Conditions:
1. Electron 14.0.0
2. electron-builder 22.11.7
3. electron-rebuild 3.2.3
4. serialport 9.2.1
5. node 14.15.4
6. npm 7.21.0

First step, try to build up the app:
```console
$ ./node_modules/.bin/electron-builder

• electron-builder  version=22.11.7 os=5.4.0-81-generic
  • loaded configuration  file=package.json ("build" field)
  • electron-rebuild not required if you use electron-builder, please consider to remove excess dependency from devDependencies

To ensure your native dependencies are always matched electron version, simply add script `"postinstall": "electron-builder install-app-deps" to your `package.json`
  • writing effective config  file=builds/builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=@serialport/bindings@9.2.1 platform=linux arch=x64
  • install prebuilt binary  name=@serialport/bindings version=9.2.1 platform=linux arch=x64 napi=
  • packaging       platform=linux arch=x64 electron=14.0.0 appOutDir=builds/linux-unpacked
..... etc, all ok .....
```
Second: try to run the app (or builded unpacked one) and got the error:
```console
$  ./node_modules/.bin/electron ./dist/main.js
/home/project/node_modules/electron/dist/electron ./dist/main.js: symbol lookup error: /home/project/node_modules/@serialport/bindings/build/Release/bindings.node: undefined symbol: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeEPKNS_9CFunctionE
```
Solution:
```console
$ ./node_modules/.bin/electron-rebuild -f
... compilation process ...
$ ./node_modules/.bin/electron ./dist/main.js
... all ok ...
```
If I try to `electron-builder` again, I will got the same error. I figured out that the problem is in serialport _prebuild_, exactly in `bindings-v9.2.1-electron-v89-linux-x64.tar.gz` that is being downloaded by electron-builder. I managed to successfully build my app only after following steps, to prevent _prebuild binaries_ being downloaded:
```console
$ ./node_modules/.bin/electron-rebuild -f
$ ./node_modules/.bin/electron-builder -c.npmRebuild=false
```
After downgrading Electron to 13.3.0, all works as expected: `electron-builder` downloads precompiled binaries (v89) and they work okay, nothing breaks.

So, is it okay that `bindings-v9.2.1-electron-v89` does not work with Electron v14.0.0?

frank-pian, 2021-09-07:
It took me a whole day

reconbot, 2021-09-07:
It looks like the latest version of electron has disabled some of the apis that we use (and that still work in node) we have a draft pr where we're working on shifting to the newer apis. In the meantime the latest version of electron won't work.

GazHank, 2021-09-08:
I can confirm this is related to the removal of `app.allowRendererProcessReuse=false` option as a breaking change in Electron 14, so reverting to version 13 is required for now.  It has been deprecated for some time, but it's final removal is what caused the breakage.

The draft change (https://github.com/serialport/node-serialport/pull/2305) is currently testable on Windows, but requires further work to complete on Linux and Mac.  I've just run a few tests on the Windows build and it functions correctly on Electron 14, so hopefully we can get the mac and linux side of the change completed soon.

GazHank, 2021-09-29:
A draft PR (https://github.com/serialport/node-serialport/pull/2305) to allow operation on Electron 14+ is now testable on all supported platforms.  

Please do give it a try if you are able, and let us know if you find any bugs or issues.  This is a significant change to the inner workings of the package; so we really need everyone's help to test it, to make sure we can iron out any issues before we set this live

1pxone, 2021-11-13:
@reconbot Hey! :) I pushed some workarounds to fix this problem. Tested it with 14 and 16 node.js and 15 electron. 

Here is my pr #2352

1pxone, 2021-11-15:
@pungfen you can fix it locally as I did in #2352 until it merged

reconbot, 2021-11-26:
https://github.com/serialport/node-serialport/pull/2360 fixes ABI issues for electron 14+ released in @serialport/bindings@9.2.8

reconbot, 2021-12-11:
I've just released `serialport@10.0.0` with NAPI support under the beta release tag. (npm i serialport@beta will get it for you.) We're going to stick with beta releases until we get some feedback on the stability. Please try it out!

Vegvizer, 2022-02-13:
how can i use serialport with node 14.19.0?
seems it doesn't take it correctly as native module

reconbot, 2022-02-14:
`serialport@10.3.0` has been released (out of beta!) you can find the upgrade guide in the docs https://serialport.io/docs/guide-upgrade#upgrading-from-9x-to-10x