serialport/node-serialport #2321: Can't solve issue with bindings file (non-election related)

Geekz45679, 2021-09-24:
I'm trying to the [serialport package[(https://www.npmjs.com/package/serialport) to communicate with an Arduino, but I can't seem to resolve an error that I'm having with my install of serialport.

**I am running:**
- Windows 11 (21H2 22000.194),
- Node: 14.17.6 LTS
- npm: 7.24.1
- serialport: 9.2.3

**Error log:**
```
C:\Users\redacted\redacted\node_modules\bindings\bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\build\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\build\Debug\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\build\Release\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\out\Debug\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\Debug\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\out\Release\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\Release\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\build\default\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\compiled\14.17.6\win32\x64\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\addon-build\release\install-root\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\addon-build\debug\install-root\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\addon-build\default\install-root\bindings.node
 → C:\Users\redacted\redacted\node_modules\@serialport\bindings\lib\binding\node-v83-win32-x64\bindings.node
    at bindings (C:\Users\redacted\redacted\node_modules\bindings\bindings.js:126:9)
    at Object.<anonymous> (C:\Users\redacted\redacted\node_modules\@serialport\bindings\lib\win32.js:1:36)
    at Module._compile (internal/modules/cjs/loader.js:1072:14)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1101:10)
    at Module.load (internal/modules/cjs/loader.js:937:32)
    at Function.Module._load (internal/modules/cjs/loader.js:778:12)
    at Module.require (internal/modules/cjs/loader.js:961:19)
    at require (internal/modules/cjs/helpers.js:92:18)
    at Object.<anonymous> (C:\Users\redacted\redacted\node_modules\@serialport\bindings\lib\index.js:6:22)
    at Module._compile (internal/modules/cjs/loader.js:1072:14) {
  tries: [
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\build\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\build\\Debug\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\build\\Release\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\out\\Debug\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\Debug\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\out\\Release\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\Release\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\build\\default\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\compiled\\14.17.6\\win32\\x64\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\addon-build\\release\\install-root\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\addon-build\\debug\\install-root\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\addon-build\\default\\install-root\\bindings.node',
    'C:\\Users\\redacted\\redacted\\node_modules\\@serialport\\bindings\\lib\\binding\\node-v83-win32-x64\\bindings.node'
  ]
}
```
**So far I've tried:**
1) Using nvm to switch to the long term supported version of Node.js (v14.17.6)
2) Reinstalling the build tools through the Visual Studio installer
3) Using `npm rebuild`
4) Reinstalling Python
5) Reinstalling the module several times

I just can't seem to get it to work. Does anyone have any ideas that I could try?

GazHank, 2021-09-24:
Hi @Geekz45679 could you please confirm the versions of VS build tools and python that you are using?

Geekz45679, 2021-09-24:
> Hi @Geekz45679 could you please confirm the versions of VS build tools and python that you are using?

[image]

GazHank, 2021-09-24:
Thanks @Geekz45679

Could you check if you have the VS build tools included in your Path environment settings?  In my case while I have both 2019 and 2017 installed only 2017 is included in the path, so only that will work for the build process

[image]

If you have made config changes and reinstalls, did you try wiping your node_modules directory and performing a fresh npm install to make sure it isnt just leaving an earlier broken version of the package?

Geekz45679, 2021-09-24:
@GazHank Unfortunately that didn't seem to do the trick.
Here is how I have it set up:
[image]

GazHank, 2021-09-24:
> @GazHank Unfortunately that didn't seem to do the trick.
> 
> Here is how I have it set up:
> 
> [image]
> 
> 

Hmm, since you have python v2 in your path can you check it isnt trying to use that version for the build?

Also, are you using the --build-from-source option?

ed-val, 2022-05-14:
It took me a while to solve this one out. Although our cases are not identical, I had the same issue as it is a non-electron related issue. When I first installed serialport, npm warns about the node-gyp 9.0.0 library having compatibility issues with the node I was using at the time, and it also gives you compatible versions suggestions. (I was not able to replicate this warnings and get you a screenshot, sorry).

I only had this problem on Windows so I followed the node-gyp installation guide for windows[https://github.com/nodejs/node-gyp#installation](url) then use nvm-windows[https://github.com/coreybutler/nvm-windows](url) to install a node compatible version (14.13.0) then changed it and restart the node command prompt.

Hope this helps someone. 

Using:
Windows 10 Pro (19041.388)
Node: 14.4.0 LTS
npm: 6.14.5
serialport: 9.2.0

closed 2023-04-05 by Geekz45679