serialport/node-serialport #2322: Getting build errors on Windows, Node 16.10

AlCalzone, 2021-09-27:
Since Node 16.10 got released, my Github action runs are failing with the following error:
```
npm ERR! path C:\Users\RUNNER~1\AppData\Local\Temp\test-iobroker.zwave2\node_modules\@serialport\bindings
npm ERR! command failed
npm ERR! command C:\Windows\system32\cmd.exe /d /s /c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! prebuild-install WARN install No prebuilt binaries found (target=16.10.0 runtime=node arch=x64 libc= platform=win32)
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@16.10.0 | win32 | x64
npm ERR! gyp info find Python using Python version 3.7.9 found at "C:\hostedtoolcache\windows\Python\3.7.9\x64\python3.exe"
npm ERR! gyp http GET nodejs.org/download/release/v16.10.0/node-v16.10.0-headers.tar.gz
npm ERR! gyp http 200 nodejs.org/download/release/v16.10.0/node-v16.10.0-headers.tar.gz
npm ERR! gyp http GET nodejs.org/download/release/v16.10.0/SHASUMS256.txt
npm ERR! gyp http GET nodejs.org/download/release/v16.10.0/win-x86/node.lib
npm ERR! gyp http GET nodejs.org/download/release/v16.10.0/win-arm64/node.lib
npm ERR! gyp http GET nodejs.org/download/release/v16.10.0/win-x64/node.lib
npm ERR! gyp http 200 nodejs.org/download/release/v16.10.0/win-x86/node.lib
npm ERR! gyp http 200 nodejs.org/download/release/v16.10.0/SHASUMS256.txt
npm ERR! gyp http 404 nodejs.org/download/release/v16.10.0/win-arm64/node.lib
npm ERR! gyp http 200 nodejs.org/download/release/v16.10.0/win-x64/node.lib
npm ERR! gyp info find VS using VS2019 (16.11.31702.278) found at:
npm ERR! gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise"
npm ERR! gyp info find VS run with --verbose for detailed information
npm ERR! (node:920) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
npm ERR! (Use `node --trace-deprecation ...` to show where the warning was created)
npm ERR! gyp info spawn C:\hostedtoolcache\windows\Python\3.7.9\x64\python3.exe
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   'C:\\hostedtoolcache\\windows\\node\\16.10.0\\x64\\node_modules\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'msvs',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\RUNNER~1\\AppData\\Local\\Temp\\test-iobroker.zwave2\\node_modules\\@serialport\\bindings\\build\\config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\hostedtoolcache\\windows\\node\\16.10.0\\x64\\node_modules\\npm\\node_modules\\node-gyp\\addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   'C:\\Users\\runneradmin\\AppData\\Local\\node-gyp\\Cache\\16.10.0\\common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=C:\\Users\\runneradmin\\AppData\\Local\\node-gyp\\Cache\\16.10.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=C:\\hostedtoolcache\\windows\\node\\16.10.0\\x64\\node_modules\\npm\\node_modules\\node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\runneradmin\\\\AppData\\\\Local\\\\node-gyp\\\\Cache\\\\16.10.0\\\\<(target_arch)\\\\node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\RUNNER~1\\AppData\\Local\\Temp\\test-iobroker.zwave2\\node_modules\\@serialport\\bindings',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'C:\\Users\\RUNNER~1\\AppData\\Local\\Temp\\test-iobroker.zwave2\\node_modules\\@serialport\\bindings\\build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp: C:\Users\runneradmin\AppData\Local\node-gyp\Cache\16.10.0\common.gypi not found (cwd: C:\Users\RUNNER~1\AppData\Local\Temp\test-iobroker.zwave2\node_modules\@serialport\bindings) while reading includes of binding.gyp while trying to load binding.gyp
npm ERR! gyp ERR! configure error 
npm ERR! gyp ERR! stack Error: `gyp` failed with exit code: 1
npm ERR! gyp ERR! stack     at ChildProcess.onCpExit (C:\hostedtoolcache\windows\node\16.10.0\x64\node_modules\npm\node_modules\node-gyp\lib\configure.js:351:16)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:390:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
npm ERR! gyp ERR! System Windows_NT 10.0.17763
npm ERR! gyp ERR! command "C:\\hostedtoolcache\\windows\\node\\16.10.0\\x64\\node.exe" "C:\\hostedtoolcache\\windows\\node\\16.10.0\\x64\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd C:\Users\RUNNER~1\AppData\Local\Temp\test-iobroker.zwave2\node_modules\@serialport\bindings
npm ERR! gyp ERR! node -v v16.10.0
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok
```

I'm not sure whether this is due to something funky on Github Actions or if there's something in `serialport` that needs to be changed.

GazHank, 2021-09-28:
Hi @AlCalzone are you still facing issues?  

I've tried building https://github.com/AlCalzone/ioBroker.zwave2.git locally and can't see any issues with 16.10.0

I'm not sure if I'm looking at the wrong repo, or if there is somethig else about my build environment which means the issue isnt appearing

AlCalzone, 2021-09-28:
I can confirm that it works locally on 16.10. Even on GitHub, it seems to be intermittent with no real logic to it:
ok: https://github.com/AlCalzone/ioBroker.zwave2/actions/runs/1278742924
nok: https://github.com/AlCalzone/ioBroker.zwave2/actions/runs/1278746979
ok again: https://github.com/AlCalzone/ioBroker.zwave2/actions/runs/1279866473

About half the builds work, the other half don't.

GazHank, 2021-09-28:
I wondered if was related that the GitHub Action environment was updated around the same as you saw the issues...  https://github.com/actions/virtual-environments/commits/main/images/win/Windows2019-Readme.md

However it looks like successful and failed run both use Version: 20210920.1

If there is any discrepancy you can see between the runs, then please let me know.

AlCalzone, 2022-04-16:
I haven't seen this since, so it probably was caused by the build system changes.

closed 2022-04-16 by AlCalzone