serialport/node-serialport #2339: SerialPort.list() crashes Node on Windows while connecting and disconnecting devices.

gabrieleimotion, 2021-11-02:
## 💥  BUG

I've experienced the issue where I have a QRCode reader attached to the usb port of my machine, this device goes to stand-by after some time and it gets back when used. 
If I query for the device lists in this exact moment the nodejs process crashes without any unhandled exception on promise. 

How can I debug this?

GazHank, 2021-11-02:
Hi @gabrieleimotion are you able to confirm the node and SP versions you are using?

There was a recent change to the windows list function which improved stability, so if you are not already using the latest version then it might be worth a try

gabrieleimotion, 2021-11-02:
@GazHank fair question, I updated my issue adding the version info :)

GazHank, 2021-11-02:
Are you able to upgrade to v12+ of node since the prior versions are no longer supported

gabrieleimotion, 2021-11-05:
I tried with v12.22.6 and still crashing without any error on serialport list.

GazHank, 2021-11-06:
If you are still facing issues with a brother printer on the latest version then perhaps https://github.com/serialport/node-serialport/issues/2259 isn't fully resolved yet :-(

gabrieleimotion, 2021-11-08:
Yes that's what I thought, but this happens sporadically and only on a combination of factors.
Is there useful a way I can get more info about the crash?

closed 2022-04-28 by stale[bot]