serialport/node-serialport #2344: Windows install does not download binaries

dzarda, 2021-11-06:
Calling `npm install serialport` on Windows 10 64-bit does not download the prebuilt binaries. Instead it tries to build from source which fails with the following message when the correct Visual Studio machinery is not installed:

```
> npm install serialport
npm ERR! code 1
npm ERR! path C:\Users\Dzarda\Documents\src\Jaculus\node_modules\@serialport\bindings
npm ERR! command failed
npm ERR! command C:\Windows\system32\cmd.exe /d /s /c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! prebuild-install WARN install No prebuilt binaries found (target=16.11.0 runtime=node arch=x64 libc= platform=win32)
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.2.0
npm ERR! gyp info using node@16.11.0 | win32 | x64
npm ERR! gyp info find Python using Python version 3.9.6 found at "C:\Python39\python.exe"
npm ERR! gyp http GET https://nodejs.org/download/release/v16.11.0/node-v16.11.0-headers.tar.gz
npm ERR! gyp http 200 https://nodejs.org/download/release/v16.11.0/node-v16.11.0-headers.tar.gz
npm ERR! gyp http GET https://nodejs.org/download/release/v16.11.0/SHASUMS256.txt
npm ERR! gyp http GET https://nodejs.org/download/release/v16.11.0/win-x86/node.lib
npm ERR! gyp http GET https://nodejs.org/download/release/v16.11.0/win-x64/node.lib
npm ERR! gyp http GET https://nodejs.org/download/release/v16.11.0/win-arm64/node.lib
npm ERR! gyp http 200 https://nodejs.org/download/release/v16.11.0/SHASUMS256.txt
npm ERR! gyp http 200 https://nodejs.org/download/release/v16.11.0/win-x64/node.lib
npm ERR! gyp http 404 https://nodejs.org/download/release/v16.11.0/win-arm64/node.lib
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS msvs_version not set from command line or npm config
npm ERR! gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
npm ERR! gyp ERR! find VS could not use PowerShell to find Visual Studio 2017 or newer, try re-running with '--loglevel silly' for more details
npm ERR! gyp ERR! find VS looking for Visual Studio 2015
npm ERR! gyp ERR! find VS - not found
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS You need to install the latest version of Visual Studio
npm ERR! gyp ERR! find VS including the "Desktop development with C++" workload.
npm ERR! gyp ERR! find VS For more information consult the documentation at:
npm ERR! gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
npm ERR! gyp ERR! find VS **************************************************************
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack Error: Could not find any Visual Studio installation to use
npm ERR! gyp ERR! stack     at VisualStudioFinder.fail (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
npm ERR! gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16
npm ERR! gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:351:14)
npm ERR! gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:70:14
npm ERR! gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16
npm ERR! gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
npm ERR! gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
npm ERR! gyp ERR! stack     at ChildProcess.exithandler (node:child_process:404:5)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:390:28)
npm ERR! gyp ERR! stack     at maybeClose (node:internal/child_process:1064:16)
npm ERR! gyp ERR! System Windows_NT 10.0.19043
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd C:\Users\Dzarda\Documents\src\Jaculus\node_modules\@serialport\bindings
npm ERR! gyp ERR! node -v v16.11.0
npm ERR! gyp ERR! node-gyp -v v8.2.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Dzarda\AppData\Local\npm-cache\_logs\2021-11-05T23_58_41_948Z-debug.log
```

[2021-11-05T23_58_41_948Z-debug.log](https://github.com/serialport/node-serialport/files/7489411/2021-11-05T23_58_41_948Z-debug.log)

This happened on two machines, both 64-bit:
- tomorrow when I tried installing it on my laptop where it was never installed
- tonight on my PC where it had worked previously but the configuration had changed somehow and `serialport` had disappeared. I think Node wanted to regenerate something idk. I then ran `npm install serialport` which led to this report.

GazHank, 2021-11-06:
Could you check the version of visual studio build tools you have installed and included within the path environment variable?

dzarda, 2021-11-06:
I probably installed Build Tools 2017 yesterday using ` npm install --global windows-build-tools` and installation passes now.

GazHank, 2021-11-06:
let me know if you face any more issues

dzarda, 2021-11-06:
I mean the issue still exists, as far as I see it. Isn't it supposed to install binaries on 64b Windows like it used to - without requiring the VS tools?

pedro-w, 2021-11-08:
I too would prefer to use serialport on a Windows machine without having to install the VS tools. The current version on npm.js is [9.2.5](https://www.npmjs.com/package/serialport/v/9.2.5) but the latest binary release on github is [9.2.4](https://github.com/serialport/node-serialport/releases/tag/%40serialport%2Fbindings%409.2.4). Usually there is only a short delay / no delay before the versions are brought in sync. Is it still the plan to provide a 9.2.5 binary release on GitHub?
Thanks for your help.

GazHank, 2021-11-08:
Oh, I wasn't aware that v 9.2.5 of the bindings hadn't been published to github, that probably explains why https://github.com/serialport/node-serialport/issues/2339 is still occuring too.

@reconbot are you able to publish the v9.2.5 version of the bindings?

pedro-w, 2021-11-08:
Looks to me like the node 17 build fails because of missing header file `v8-weak-callback-info.h` ([here](https://github.com/serialport/node-serialport/runs/4138697251?check_suite_focus=true#step:5:5617)), which may well be fixed in https://github.com/nodejs/node/pull/40526 (released as node 17.0.1) Does that look like a plausible explanation?

reconbot, 2021-11-08:
Prebuild doesn't have 17.0.1 yet via node-abi https://github.com/electron/node-abi

reconbot, 2021-11-08:
Prebuild doesn't have 17.0.1 yet via node-abi https://github.com/electron/node-abi which gets it's version info from https://cdn.jsdelivr.net/gh/nodejs/node/doc/abi_version_registry.json which doesn't have 17.0.1 yet. It doesn't look like it will however, so we need to figure out where prebuild is getting which version to download...

I opened an issue there https://github.com/prebuild/prebuild/issues/284

GazHank, 2021-11-08:
Is the situation any better with prebuildify?  I've started work on switching to prebuildify post N-API and it looks pretty good so far; at the moment I'm trying to work out if V10 should be only a switch to N-API or also switch to prebuildify at the same time...

reconbot, 2021-11-09:
It would be the same deal.

jakobrosenberg, 2021-11-11:
@reconbot is there anything I can do to contribute towards `prebuildify` integration?

GazHank, 2021-11-11:
> @reconbot is there anything I can do to contribute towards `prebuildify` integration?

I have a draft prebuildify branch which I've not yet pushed to GitHub, which builds upon the N-API migration code.  I'll try to share this once I get back to my dev machine.  I'm still not sure if v10 should initially focus on the switch to N-API or if we should include prebuildify too; perhaps if I share that code and get some feedback it will help make that decision

jakobrosenberg, 2021-11-11:
@GazHank that sounds great! I'll be happy to give you feedback tomorrow.

jakobrosenberg, 2021-11-15:
@GazHank is this the branch you mentioned? https://github.com/GazHank/node-serialport/tree/napi

GazHank, 2021-11-15:
Hi @jakobrosenberg thats the node-api migration branch (which is also viewable as a PR).  I haven't had chance to publish the prebuildify branch yet; it is dependent up the node-api migration but a far smaller change

reconbot, 2021-11-18:
looks like the builds are fixed in `node-abi@3.5.0`, I've updated and am publishing `serialport@9.2.7` now 🤞

reconbot, 2021-11-18:
I had to remove node 17 builds but we're back in business

intensite, 2021-12-03:
So is it possible to install prebuilt libraries for Windows with:
$ node --version
v17.1.0