serialport/node-serialport #2346: [Bug] bindings-v9.2.4-electron-v80-win32-x64 crashes Atom plugin

jakobrosenberg, 2021-11-08:
First off, thanks for this amazing library. I've been using it for a while, but have run into issues when using it in Atom.

To reproduce
1. Create new Atom package
2. npm install serialport
3. Add `import 'serialport'` to `lib/my-package.js`
4. Download and unzip `bindings.node` from [release](https://github.com/serialport/node-serialport/releases/download/%40serialport%2Fbindings%409.2.4/bindings-v9.2.4-electron-v80-win32-x64.tar.gz) to `node_modules/@serialport/bindings/build/Release/bindings.node`
5. Crash

[image]

```
node module version: 80 // from console.log(process.versions.modules) in package
serialport: 9.2.4
@serialport/bindings: 9.2.5
```

Fingers crossed I'm doing something wrong.

Ps. the issues template supports proposal and documentation requests, but not reporting of bugs. Is this intentional?

reconbot, 2021-11-08:
The bindings should automatically download in most cases. However the 9.2.5 bindings were missing and should be up now. Atom isn't a supported environment so you may have to do it's equivalent of `electron-rebuild` to get it working.

jakobrosenberg, 2021-11-08:
Thanks for the quick reply.

How are bindings downloaded in an Electron environment? Is it possible? Currently I'm using a customized version of `prebuild-install` to detect the module version inside of Electron. I then grab the bindings with `prebuild-install's `download` function. It's not pretty but for VSCode it works. Alas Atom is another story.

reconbot, 2021-11-18:
`serialport@9.2.7` is being built, might help.

npm post install installs the `.node` file from whatever is running npm, that's usually node. That's why we need electron-rebuild. You can download from the [release](https://github.com/serialport/node-serialport/releases) files and put them in your build folder in the binding's package.

jakobrosenberg, 2021-11-29:
Hi @reconbot completely missed your reply. Sorry!

I had a look at the releases, but there are no v87 versions which is needed for Atom.

Eg. for Windows, you would need this file to use `serialport` i Atom.
https://github.com/serialport/node-serialport/rele%E2%80%A62.8/bindings-v9.2.8-electron-v87-win32-x64.tar.gz

reconbot, 2022-02-14:
`serialport@10` is out which uses n-api and should work without issue. https://serialport.io/docs/guide-upgrade#upgrading-from-9x-to-10x

closed 2022-02-14 by reconbot