serialport/node-serialport #2351: "close" event not triggered when disconnecting device

fur-gan, 2021-11-13:
### SerialPort Version

9.0.8

### Node Version

v12.14.1

### Platform

Microsoft Windows NT 10.0.19043.0 x64

### Architecture

x64

### What steps will reproduce the bug?

```
const Serialport = require("serialport");

Serialport.list()
  .then((ports) => {
    if (ports.length > 0) {
      const port = new Serialport(ports[0].path, { autoOpen: true });
      port.on("open", () => {
        console.log("Port is opened.");
      });
      port.on("close", (error) => {
        console.log("Port is closed: ", error.message);
      });
      setInterval(() => {
        port.write("main screen turn on", (error) => {
          if (error) {
            return console.log("Error on write: ", error.message);
          }
          console.log("message written");
        });
      }, 10000);
    }
  })
  .catch((error) => {
    console.log("Error: " + error.message);
  });
```

### What happens?

Detect whenever a serial device is disconnected. I try to achieve this with the `"close"` event but it doesn't get triggered when I disconnect/pull out the device from my computer. Also tried it with virtual serial ports but same result.

### What should have happened?

Whenever a serial device is disconnected from the computer, the event `"close"` should be triggered.

### Additional information

I've also tried it with the version `9.2.5` but same result, the expected behaviour does not happen.

GazHank, 2021-11-13:
I believe that the close event should trigger if you try to interact with the port (read/write); but if you simply unplug and don't try to interact, then the close event may not be triggered.

fur-gan, 2021-11-18:
Yeah, I noticed it too that the event gets triggered when I tried to write to the device. I thought it would be triggered as soon as I unplug the device that's why I was confused.

I guess a workaround for this would be detecting all devices periodically.

Thank you for clarifying.

closed 2021-11-18 by fur-gan

GazHank, 2021-11-19:
We should probably update the docs to clarify the behaviour

reconbot, 2021-11-19:
If it's not there already 👍

GazHank, 2021-11-19:
> If it's not there already 👍

I couldn't see it in the docs with a quick scan; but I'm stuck on mobile/phone app at the moment; so that will be my excuse if I have missed it 😳

fur-gan, 2021-11-19:
In the [docs](https://serialport.io/docs/api-stream#events) it's written as 
_The close event is emitted when the port is closed. In the case of a disconnect, it will be called with a Disconnect Error object (err.disconnected == true). In the event of an error while closing (unlikely), an error event is triggered._

At least to me it wasn't very clear if the `close` event would be called when we unplug the device

Anways thanks a lot for taking a look into this issue and thank you for your great work with this library

NbrTDB, 2024-11-12:
I don't suppose there's any way to get the old behaviour back? I was relying on that to detect when people ripped out USB serial devices and handle that appropriately. If not does anyone have a good alternative other than spamming traffic at the port?