serialport/node-serialport #2353: port.on("data") event slow when send speed is less than 500ms

wmt-php-Nikhil, 2021-11-16:
### SerialPort Version

^9.2.4

### Node Version

v17.0.1

### Electron Version

13.2.1

### What steps will reproduce the bug?

After opening the port and starting communication. when we send data in intervals less than 500ms we the port.on("data") data receive is delayed bs 2-3 sec. resulting in poor performance.

### What happens?

data is received after 2-3s.

### What should have happened?

data should be received in realtime

GazHank, 2021-11-16:
Can I check which OS you are using?  The logic is quite discrete for windows vs the other platforms, so this info might help troubleshooting.  Also if on Linux can you confirm if you are using the Low Latency feature?

wmt-php-Nikhil, 2021-11-17:
I am using windows 10 but the application we are building should be supported for windows 7 onwards. Let me know if there is a way for low latency on windows

neelv-wmt, 2021-11-18:
@GazHank Any update on this. How can I resolve the latency issue

wmt-php-Nikhil, 2021-11-18:
@GazHank Any updates. How can I resolve the issue

reconbot, 2021-11-18:
I'd share some code, try the debugging steps from the docs and check your hardware. You haven't really given us a lot to go on.

closed 2022-04-28 by stale[bot]