serialport/node-serialport #2357: Cannot install serialport on node v12.22.7

CyberHippo, 2021-11-19:
### SerialPort Version

9.2.7

### Node Version

v12.22.7

### Electron Version

/

### Platform

Linux 412a74f34ede 5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64 Linux

### Architecture

x86_64

### Hardware or chipset of serialport

/

### What steps will reproduce the bug?

Install serialport as follows :
```Dockerfile
RUN apk add --no-cache nodejs make gcc g++ python3 linux-headers udev
RUN npm install -g serialport --build-from-source  --unsafe-perm
```

### What happens?

The installation does not succeed.  Please see the attached logs:

```
> @serialport/bindings@9.2.7 install /usr/app/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
prebuild-install WARN install No prebuilt binaries found (target=12.22.7 runtime=node arch=x64 libc=musl platform=linux)
make: Entering directory '/usr/app/node_modules/@serialport/bindings/build'
make: Leaving directory '/usr/app/node_modules/@serialport/bindings/build'
make: printf: Operation not permitted
make: *** [bindings.target.mk:114: Release/obj.target/bindings/src/serialport.o] Error 127
build error
ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:314:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Linux 4.9.0-11-amd64
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/app/node_modules/@serialport/bindings
gyp ERR! node -v v12.22.7
```

### What should have happened?

The package is installed from source successfully.

### Additional information

Let me know if you need additional information.

ws02589111, 2021-11-22:
```
error /app/node_modules/@serialport/bindings: Command failed.
Exit code: 1
Command: prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
Arguments: 
Directory: /app/node_modules/@serialport/bindings
Output:
prebuild-install WARN install No prebuilt binaries found (target=12.22.7 runtime=node arch=x64 libc=musl platform=linux)
gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@12.22.7 | linux | x64
gyp info find Python using Python version 2.7.18 found at "/usr/bin/python"
gyp http GET https://unofficial-builds.nodejs.org/download/release/v12.22.7/node-v12.22.7-headers.tar.gz
gyp http 200 https://unofficial-builds.nodejs.org/download/release/v12.22.7/node-v12.22.7-headers.tar.gz
gyp http GET https://unofficial-builds.nodejs.org/download/release/v12.22.7/SHASUMS256.txt
gyp http 200 https://unofficial-builds.nodejs.org/download/release/v12.22.7/SHASUMS256.txt
gyp info spawn /usr/bin/python
gyp info spawn args [
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/app/node_modules/@serialport/bindings/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.cache/node-gyp/12.22.7/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/12.22.7',
gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/12.22.7/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/app/node_modules/@serialport/bindings',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/app/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:78:51: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
   78 |   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
      |                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:135:53: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  135 |   uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:175:52: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  175 |   uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:215:52: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  215 |   uv_queue_work(uv_default_loop(), req, EIO_Flush, (uv_after_work_cb)EIO_AfterFlush);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:271:50: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  271 |   uv_queue_work(uv_default_loop(), req, EIO_Set, (uv_after_work_cb)EIO_AfterSet);
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Get(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:316:50: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type]
  316 |   uv_queue_work(uv_default_loop(), req, EIO_Get, (uv_after_work_cb)EIO_AfterGet);
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE GetBaudRate(Nan::NAN_METHOD_ARGS_TYPE)':
[... cut ...]
gyp ERR! stack     at ChildProcess.emit (events.js:314:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Linux 5.8.0-63-generic
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /app/node_modules/@serialport/bindings
gyp ERR! node -v v12.22.7
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.
```
Same issue get

reconbot, 2022-02-05:
Want to give our latest beta a try? it has more prebuilt binaries. `serialport@10.2.2`