serialport/node-serialport #2384: Move to @serialport/bindings-cpp

reconbot, 2022-01-07:
This issue is to track moving `bindings` out of the monorepo into [serialport/bindings-cpp](https://github.com/serialport/bindings-cpp) and renaming it. The reasons for the move are scattered over a few prs and issues so I'll collect them here.

1. We'd like to ship the binaries in the npm package. 
2. We don't want all maintainer's laptops to have "production" access
3. Having an automated publishing pipeline with binary building is difficult or impossible with lerna and all the other packages. (I can count 3 failed attempts on my part.)

Since we're about to move to v10 which is currently in beta for napi testing (`serialport@beta` lead by @GazHank) it's the perfect time to make these changes.

I've created `serialport/bindings-cpp` and it needs the following;
- [x] code transferred over to the new repo preserving history
- [x]  [`semantic-release`](https://semantic-release.gitbook.io/semantic-release/) setup in a build pipeline with `prebuild-cross` (much of this work was completed on the main repo by @thegecko)
- [ ] Switch v10 to use the new package
- [ ] Removal of the bindings package in the monorepo
- [ ] v10 Docs to be updated with new package names and repo locations

After v10 leaves beta
- [ ] republish `@serialport/bindings` with a new version and deprecate it so anyone using it directly gets a message about the name change to `@serialport/bindings-cpp`
- [ ] cut v10 docs

It's worth noting I've renamed the package because I'd like us to be able to have alternative bindings such as `@serialport/bindings-web` and `@serialport/bindings-rust` both projects myself and others have been exploring. 

Hope that covers everything. As always I'm open to comment and feedback ❤️

reconbot, 2022-01-07:
Worth noting that `@serialport/bindings-cpp` is available right now from npm and is build with NAPI for the following platforms via `prebuildify-cross`

    centos7-devtoolset7
    alpine
    linux-armv6
    linux-armv7
    linux-arm64
    android-armv7
    android-arm64

reconbot, 2022-01-23:
serialport@10.1.0 is now using typescript

reconbot, 2022-02-03:
`serialport@10.2.1` is now completely typescript powered and has quite different exports. https://github.com/serialport/node-serialport/pull/2406

reconbot, 2022-02-05:
`serialport@10.2.2` has fixed stopbits types and has `binding-mock` spun out to it's own package.

reconbot, 2022-02-13:
got decent bindings docs https://serialport.io/docs/next/api-bindings-cpp

reconbot, 2022-02-14:
`serialport@10` is now released 🎉

s00d, 2022-02-16:
hello, great news, but I have a problem understanding Testing
The Upgrade Guide documentation says
> SerialPort.Bindings is now read only and bindings is no longer part of the options object. 

but [here](https://serialport.io/docs/guide-testing) i see an old example
```
SerialPort.Binding = MockBinding
```

How now to implement such a variant of work? Can you update the documentation?

reconbot, 2022-02-16:
That's a mistake, I'd have a look here in a meantime https://serialport.io/docs/api-serialport#serialportmock

reconbot, 2022-02-19:
Published the deprecate packages with deprecation warnings. So far so good 👍

closed 2022-02-19 by reconbot

reconbot, 2022-02-19:
@s00d fixed in https://github.com/serialport/website/pull/84 thank you 🙏