serialport/node-serialport #2432: Unable to access MockPortBinding in serialport 10

AlCalzone, 2022-02-17:
### SerialPort Version

10.3.0

### Node Version

16.10.0

### Electron Version

-

### Platform

Microsoft Windows NT 10.0.19043.0 x64

### Architecture

x64

### Hardware or chipset of serialport

doesn't matter

### What steps will reproduce the bug?

Reproduce example for mocking from docs -> https://serialport.io/docs/api-serialport#port-1

```ts
import { SerialPortMock } from "serialport";

SerialPortMock.binding.createPort('/dev/robot')
const port = new SerialPortMock({ path: '/dev/robot', baudRate: 9600 })
port.port.emitData('data')
//    ^ port.port is undefined
```

### What happens?

Can't access the mock port, no way to `emitData` or access `lastWrite`:
```
port.port.emitData("data");
          ^

TypeError: Cannot read properties of undefined (reading 'emitData')
```

### What should have happened?

It should work, or the docs should be corrected

s00d, 2022-02-18:
Hello. Try to open the port before using. Today I finished transferring to version 10, there were no problems with this, but I decided to extend the SerialPortMock class

```ts
port.open((error?: Error | null) => {
       port.port.emitData('data')
})
```

My example

```ts
export class BaseModem extends SerialPortMock {
// ...
  sendOk() {
    this.port!.emitData("OK" + "\r\n")
    return true
  }
// ...
}

const port = new BaseModem({
          path: this.path,
          baudRate: 115200,
          autoOpen: false
})
// ...
const port.open((error?: Error | null) => {
        if (error) {
          reject(error)
        }
//....
        port.sendOk()
//....
 })
```

AlCalzone, 2022-02-18:
Perfect, that was it. I wasn't even able to find it with the debugger, but after opening the port, the property exists.

reconbot, 2022-02-19:
That's correct there is no binding until after it's been opened. Can't emit anything from a closed port. I've fixed the bad example in the docs.

closed 2022-02-19 by reconbot

AlCalzone, 2022-02-20:
> Let me know if there's anywhere else that doesn't make sense.

The entire documentation around mocking and the correct types of what we're working with was a bit complicated to navigate through. Starting with the upgrade guide, I had to click around the docs and source code quite a bit to figure out what is what when it comes to mocking.

reconbot, 2022-03-06:
I edited it a bit. Open to suggestions.