serialport/node-serialport #2433: TypeError: SerialPort is not a constructor

khunpoum, 2022-02-18:
### SerialPort Version

10.3.0

### Node Version

12.22.10

### Platform

Ubuntu 20.04.3 LTS (5.4.0-99-generic)

### Architecture

x64

### What steps will reproduce the bug?

code test.js
```
const SerialPort = require('serialport');
const port = new SerialPort({path: '/dev/ttyUSB0', baudRate: 9600});
```

npm install serialport
  serialport@10.3.0 installed

### What happens?

node test.js output:
```
const port = new SerialPort({path: '/dev/ttyUSB0', baudRate: 9600});
             ^

TypeError: SerialPort is not a constructor
    at Object.<anonymous> (/root/watchtemp/test-sp.js:2:14)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
    at internal/main/run_main_module.js:17:47

```

### What should have happened?

changed node version from 6.x to 12.x

khunpoum, 2022-02-18:
this worked only in configuration with Node v10.x and serialport@6.0.4

khunpoum, 2022-02-18:
how to make serialport work with modern versions of node like node v16.x ?

reconbot, 2022-02-19:
I'd check out the [v10 upgrade guide](https://serialport.io/docs/guide-upgrade#upgrading-from-9x-to-10x)