serialport/node-serialport #2439: Electron was frozen when closing port in main thread

shenzhuxi, 2022-02-24:
### SerialPort Version

10.3.0

### Node Version

16.13.2

### Electron Version

17.0.1

### Platform

Ubuntu 20.04

### Architecture

x64

### What steps will reproduce the bug?

The main.js
```
const { SerialPort } = require('serialport')
const { ReadlineParser } = require('@serialport/parser-readline')
const GPS = require('gps');

let gpsSetting = store.get('gps');
let gpsOptions = {
  path: gpsSetting.path,
  baudRate: gpsSetting.baudRate,
  autoOpen: false
}
let port = new SerialPort(gpsOptions);
let gps = new GPS;
// IPC listener
ipcMain.on('electron-gps-get', async (event) => {
  if (port.isOpen) {
    event.returnValue = gps.state;
  }
  else {
    event.returnValue = false;
  }
});
ipcMain.on('electron-gps-open', async (event) => {
  if (port.isOpen) {
    console.log('GPS was already on.');
  }
  else {
    gpsSetting = store.get('gps');
    gpsOptions = {
      path: '/dev/ttyACM0',
      baudRate: gpsSetting.baudRate,
      autoOpen: true
    }
    port = new SerialPort(gpsOptions);
    //https://en.wikipedia.org/wiki/NMEA_0183
    let parser = port.pipe(new ReadlineParser({ delimiter: '\r\n', encoding: 'ascii' }));
    parser.on('data', function (data) {
      gps.update(data);
    });
  }
});
ipcMain.on('electron-gps-close', async (event) => {
  port.close(function(err){
    console.log(err);
  });
});
``

### What happens?

The port can be open in either main.js or renderer.js in Electron and the gps data can be got. 
```
  port.close(function(err){
    console.log(err);
  });
```
will print null and freeze the Electron app either from the main.js or called from renderer.js with ipcRenderer.sendSync('electron-gps-close').

### What should have happened?

So to work around it, I move the open, get and close code to preload.js and everything is working except 
```
  port.close(function(err){
    console.log(err);
  });
```
still printed null. 

I think it should be similar to port.open and have meaningful message (Although it's weird to have "is opening" as an error message).

I also found that port.open is much faster in preload.js than in main.js.  

### Additional information

I think https://serialport.io/docs/next/guide-electron and https://github.com/serialport/electron-serialport should be updated because contextIsolation was recommended for recent Electronjs releases https://www.electronjs.org/docs/latest/tutorial/context-isolation.

reconbot, 2022-02-26:
Please do update our example and docs with the latest recommendations. I'd have to run some tests to see what I can reproduce from your example.

shenzhuxi, 2022-02-27:
Hi @reconbot , I just added a little bit more information. 
I tried on Raspberry pi and got the same symptoms. 

I also tried two other native modules https://github.com/fivdi/i2c-bus and https://github.com/kelly/node-i2c. Both of them can only be run once or twice in the main.js then freeze the Electron app. After I patched them with https://nodejs.github.io/node-addon-examples/special-topics/context-awareness/ like https://github.com/serialport/node-serialport/pull/2352/files, both of them can be run in the preload.js without problem.

closed 2022-03-15 by shenzhuxi