serialport/node-serialport #2446: Serial port 10.3.0 slow compared to 9.2.8 on Windows

barak14, 2022-03-04:
### SerialPort Version

10.3.0

### Node Version

v12.9.0

### Electron Version

N/A

### Platform

Microsoft Windows NT 10.0.22000.0 x64

### Architecture

x64

### Hardware or chipset of serialport

Any

### What steps will reproduce the bug?

Change package.json from v9.2.8 to 10.3.0 and run npm install

### What happens?

Currently we use 9.2.8 and the performance is fine:
2022-03-04 10:23:40.003 [trace ] [core ] - serialport send (17 bytes)
2022-03-04 10:23:40.099 [trace ] [core ] - serialport received (1 bytes)
2022-03-04 10:23:40.199 [trace ] [core ] - serialport received (141 bytes)
2022-03-04 10:23:40.200 [trace ] [core ] - serialport send (1 bytes)

after upgrading to v10.3.0 this has consistently changed to:
2022-03-04 10:01:31.437 [trace ] [core ] - serialport send (17 bytes)
2022-03-04 10:01:32.161 [trace ] [core ] - serialport received (1 bytes)
2022-03-04 10:01:32.953 [trace ] [core ] - serialport received (141 bytes)
2022-03-04 10:01:32.953 [trace ] [core ] - serialport send (1 bytes)

As you can see there is now significant delays on the writes or maybe even the receiving code which was unexpected. I would expect minor variations but to go from < 100ms to > 600ms to receive the response from the device seems extreme. Is there meant to be a change in how the stream.write is called?
It's kind of difficult to provide the source code that does this here, we are simpy using the stream interface and write/data events (ie port.write data and port.on 'data') to a particular device with a proprietry communication protocol.

### What should have happened?

Performance should be unchanged - the device/serial port hardware is not changing only the serial port driver is.
This was brought to my attention by another developer on Win11 that also experienced a significant degradation in performance.
And it is consistant - changing the version (+ the slight API changes to open the port) bring this issue on instantly and reverting the version fixes it.

dtboy1995, 2023-06-21:
the same issue

GazHank, 2023-06-21:
Are you able to share any additional info, or sample code that you are using?

I've run some check on Win11, and can't replicate a slowdown.  The checks include read and write to the same device on the same windows machine, using node 12.22.7 for both checks.  The only change was comparing serialport 9.2.8 to 10.5.0 (with the very minor syntax changes between the two)

I'll try again to see if there is any additional resource usage, which could in turn be causing a showdown for some, but not noticable depending on the host machine; however if you can share any more info to help investigate that would be very welcome

GazHank, 2023-06-22:
Running some more checks and I'm not seeing any additional resource usage, so I'm afraid I'm unable to replicate your issue.  It could be that I'm unable to see the variation due to the limited granularity of the built in windows monitoring, or I'm just not testing the aspect of the package which is causing the slowdown for you...

In case it helps, I've thrown together a quick project at https://github.com/GazHank/serialport-PerfTest which does some basic read, write and parsing to test performance.  It does include some device specific logic in the messages is sends and recieves, but otherwise it is pretty generic, so should be quick and easy to modify.  I've also kept the v9 vs v10 changes to a minimum at the top of the main.js to allow a quick switch between versions, and minimise risk of accidental changes.

Do please let me know if you have any more info on the performance impact you are seeing