serialport/node-serialport #2463: binding-cpp not building Mac

Edig, 2022-03-22:
### SerialPort Version

10.4.0

### Node Version

14

### Electron Version

17.1.2

### Platform

Mac - Monterey (12.3)

### What steps will reproduce the bug?

`npx electron-builder -w` with 

```
"devDependencies": {
    "electron": "^17.1.2",
    "electron-builder": "21.2",
    "electron-builder-squirrel-windows": "^22.14.12",
    "electron-rebuild": "^3.2.7"
  },
  "dependencies": {
    "@electron/remote": "^2.0.8",
    "bootbox": "^5.5.2",
    "bootstrap": "^4.5.3",
    "electron-is-dev": "^2.0.0",
    "electron-log": "^4.3.1",
    "electron-pos-printer": "^1.2.4",
    "electron-simple-updater": "^2.0.9",
    "electron-updater": "^4.6.5",
    "install": "^0.13.0",
    "jquery": "^3.5.1",
    "jsbarcode": "^3.11.3",
    "mysql": "^2.18.1",
    "node-thermal-printer": "^4.1.2",
    "pdf-to-printer": "^5.3.0",
    "popper.js": "^1.16.1",
    "serialport": "^10.4.0"
  }```

### What happens?

When I update to `V10.4.0` it change from binding to binding-cpp and when building for windows it sticks in. 
`• rebuilding native dependencies  dependencies=@serialport/bindings-cpp@10.7.0 platform=win32 arch=x64`

### What should have happened?

Build

### Additional information

I manage to fix it, by changing ` "@serialport/bindings-cpp": "10.7.0"` to ` "@serialport/bindings": "10.0.1"` in package.json of serialport.

Note: V10.0.1 can be build, but when you run it, it says "Serialport is not a constructor" you need to upgrade minimum to 10.0.2 and after that version, the binding change to cpp

shenzhuxi, 2022-08-25:
I also have this problem with electron-builder on Raspberry Pi. 
Electron-builder has no problem to build other native modules like https://github.com/fivdi/i2c-bus with package.json like
```
  "scripts": {
    ...
    "postinstall": "electron-builder install-app-deps"
  },
```
I wonder if it's because [serialport/bindings-cpp](https://github.com/serialport/bindings-cpp/blob/main/.github/workflows/build.yml) provided prebuilts/linux-arm/ and other platforms like Mac, so electron-builder won't consider it's necessary to build it from source. 

To work around, I just "rm -f node_modules/@serialport" and "npm install serialport --build-from-source".

GazHank, 2023-06-27:
per https://github.com/serialport/node-serialport/issues/2619 cross building using electron-builder should now work correctly in electron-builder 24.5.1