serialport/node-serialport #2464: `No native build was found for platform` when using Electron Forge + Webpack

chetbox, 2022-03-25:
### SerialPort Version

10.4.0

### Node Version

14.19.1

### Electron Version

17.2.0

### Platform

Linux [redacted] 5.13.0-37-generic #42~20.04.1-Ubuntu SMP Tue Mar 15 15:44:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Architecture

x64

### What steps will reproduce the bug?

I'm having trouble using the Electron Forge Typescript + Webpack template to replicate this example: https://github.com/serialport/electron-serialport

- `yarn create electron-app my-new-app --template=typescript-webpack && cd my-new-app`
- `yarn add serialport tableify`
- `yarn add --dev @types/tableify
- Copy the contents of [`renderer.js`](https://github.com/serialport/electron-serialport/blob/master/renderer.js) into `renderer.ts`
- Copy the contents of [`index.html`](https://github.com/serialport/electron-serialport/blob/master/index.html) into `index.html`
- Add `webPreferences: { nodeIntegration: true, contextIsolation: false }` to the options of `BrowserWindow` in `index.ts`
- `yarn start`

### What happens?

This error appears on the console and Serialport fails to start.

```
Uncaught Error: No native build was found for platform=linux arch=x64 runtime=electron abi=101 uv=1 libc=glibc node=16.13.0 electron=17.2.0 webpack=true
    loaded from: [redacted]/my-new-app/node_modules/electron/dist/resources/electron.asar

    at Function.load.path (index.js?04e8:6:99)
    at load (index.js?04e8:6:99)
    at eval (load-bindings.js?bdc2:10:1)
    at Object../node_modules/@serialport/bindings-cpp/dist/load-bindings.js (index.js:85:1)
    at __webpack_require__ (index.js:841:33)
    at fn (index.js:1028:21)
    at eval (darwin.js?fd34:8:25)
    at Object../node_modules/@serialport/bindings-cpp/dist/darwin.js (index.js:30:1)
    at __webpack_require__ (index.js:841:33)
    at fn (index.js:1028:21)
```

### What should have happened?

Serialport should list all serial devices in the Electron window that opens exactly as with https://github.com/serialport/electron-serialport

### Additional information

The same issue occurs when running a packaged Electron app using `yarn make`.

I have observed the same behaviour using Electron 12.0.9. (This is the project we are using Serialport in.)

A colleague has replicated the issue with Electron 12.0.9 on macOS with an M1 MacBook Pro.

chetbox, 2022-03-25:
I think the issue boils down to compatiblity with https://github.com/vercel/webpack-asset-relocator-loader

It is looking for one of the following:

- `require('bindings')(...)`
- `nbind.init(..)`
- `node-pre-gyp` include patterns

I haven't been able to cooerce Electron to load the `.node` native libraries by copying them to `.webpack/renderer/native_modules/prebuilds/linux-x64` sadly.

PredictabilityIsGood, 2022-04-18:
@chetbox Did you ever get this working? The moment I added serialport toone of my electron projects, I encountered the same bindings issue. Before I go on a wild goose chase, I'd like to see if you were able to resolve this issue and provide some guidance

chetbox, 2022-04-18:
Not yet. We'll have to look when we absolutely must upgrade which is not yet. Chase away. Please update us if you make any progress.

mimamuh, 2022-04-30:
I got the same error adding serialport to an electron forge + webpack project.

sh0shinsha, 2022-05-10:
hey @PredictabilityIsGood @mimamuh any updates?  same experience as you guys using electron-builder

reconbot, 2022-05-10:
You don't need to builder anymore (since v10) but you do need to tell webpack to ignore it https://webpack.js.org/configuration/externals/

mimamuh, 2022-05-10:
@sh0shinsha I switched to [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate) which excludes the serialport libs from the bundler similar as @reconbot mentioned. So simply try to exclude `serialport` and I think it should work even with forge + webpack then.

chetbox, 2022-05-10:
I've tried adding `serialport` and `@serialport/bindings-cpp` to webpack's `externals` but I still get this message in the renderer after running `electron-forge make`:

```
Uncaught Error: Cannot find module 'serialport'
```

chetbox, 2022-05-10:
We're using the ASAR option to package a production app. On closer inspection it seems the ASAR has an empty `node_modules` folder. Is there anything I need to do to tell Electron Forge to include `serialport` in `node_modules` inside the ASAR?

mimamuh, 2022-05-10:
@sh0shinsha Yes, I got it managed to work with [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate). Have you installed `serialport` like desribed [here in the docs?](https://electron-react-boilerplate.js.org/docs/native-modules/). This might be your  issue.

@chetbox Actually I had the same thought: In case you exclude `serialport` from `webpacks'` build process then you might have to include the excluded module yourself into electrons build process somehow in case electron (forge) doesn't do so automatically for you, which I doubt it does [when I read the docs here.](https://www.electronforge.io/config/plugins/webpack#native-modules) That might explain to me why it's not included in your ASAR folder?

At least [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate) handles it like that. I have to install these native modules into a separated `node_modules` folder so that it can be copied it into my app before it is packaged by electron. [Check out these docs here,](https://electron-react-boilerplate.js.org/docs/native-modules/) maybe it helps to understand the process.

sh0shinsha, 2022-05-10:
@mimamuh you saved me mate!  Following the steps in the [docs you linked](https://electron-react-boilerplate.js.org/docs/native-modules/) regarding how native modules are handled in [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate) set me on the right path.  

I'll outline the steps I took to solve my `No native build was found for platform` error.  I've modeled my project's webpack configs on the ones in [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate), so these steps may be helpful for those using [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate) or those with a similar webpack setup.

1. `externals: ['serialport']` in webpack config -> in my case I'm case I just added it right to `webpack.config.main.prod` since I'm not merging configs like in [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate)
2. `library: { type: 'commonjs2' } ` in webpack config's `output` 
3. (if you've already installed `serialport`, `npm uninstall serialport` in your root `package.json`
4. `cd ./release/app`  then `npm install serialport`
- (https://electron-react-boilerplate.js.org/docs/native-modules/#native-modules-in-electron-react-boilerplate)
- (https://github.com/electron-react-boilerplate/electron-react-boilerplate/blob/main/.erb/scripts/check-native-dep.js)

If you're using  [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate) you should be all set, they handle module linking in the  `./release/app/package.json`: https://github.com/electron-react-boilerplate/electron-react-boilerplate/blob/main/release/app/package.json

In my case I set `serialport: =10.4.0` in both the `./release/app/package.json` and `devDependencies` in my root `package.json` since I'm using an npm workspace monorepo and not the exact folder structure of  [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate).  If anyone has a better solution to this I would welcome it but won't pursue it further here since it is outside the scope of `serialport`.

Hope this helps, thanks everyone!

chetbox, 2022-05-10:
> @sh0shinsha Yes, I got it managed to work with [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate). Have you installed `serialport` like desribed [...]

Those instructions for native modules with `electron-react-boilerplate` work in development using `serialport` but not when the app is packaged. I'm not sure why and I don't understand how to debug this tooling well enough. Here's a repo with the changes if anyone wants to investigate: https://github.com/chetbox/electron-react-boilerplate
(Note that I had to comment out the renderer IPC code because it caused a crash with `contextIsolation: false`.

sh0shinsha, 2022-05-10:
@chetbox it looks like `serialport` isn't present in your `./release/app/package.json`.  Let me know if the steps I outlined help.

chetbox, 2022-05-10:
@sh0shinsha yes, that's the key for Electron React Boilerplate. 👍🏼 Thanks for helping me understand how that build system works.

With Electron Forge I can get a packaged (non-ASAR) build working by copying `serialport` and its dependencies into the packaged `electron-forge` folder. 

```
yarn package
cp -r node_modules/serialport out/my-new-app-linux-x64/resources/app/node_modules/
cp -r node_modules/@serialport out/my-new-app-linux-x64/resources/app/node_modules/
cp -r node_modules/debug out/my-new-app-linux-x64/resources/app/node_modules/
cp -r node_modules/ms out/my-new-app-linux-x64/resources/app/node_modules/
cp -r node_modules/node-gyp-build out/my-new-app-linux-x64/resources/app/node_modules/
./out/my-new-app-linux-x64/my-new-app
```

I assume this means the Electron Forge + Webpack build system needs to be better aware of how to handle native modules like this. I think the issue is with [@electron-forge/plugin-webpack](https://www.npmjs.com/package/@electron-forge/plugin-webpack) or with [@vercel/webpack-asset-relocator-loader](https://github.com/vercel/webpack-asset-relocator-loader) (as I mentioned [above](https://github.com/serialport/node-serialport/issues/2464#issuecomment-1079230644)) not with `serialport`.

maneetgoyal, 2022-07-17:
Using `electron-builder`.

This worked:

```
// webpack.config.js
    externals: {
      serialport: "commonjs2 serialport", // Ref: https://copyprogramming.com/howto/electron-and-serial-ports
    },
```

PredictabilityIsGood, 2022-07-18:
@maneetgoyal Wonderful! This seems to have worked on my project as well. 

Do you know specifically why this works? Even in this linked reference, the sentence is pretty vague.

shenzhuxi, 2022-08-31:
Electron Forge is using electron-rebuild. With the simplest project which only dependents serialport, electron, and electron-rebuild without any bundle tool, It will create @serialport/bindings-cpp/build/Release/.forge-meta but won't build anything. 

>You don't need to builder anymore (since v10) but you do need to tell webpack to ignore it https://webpack.js.org/configuration/externals/
@reconbot Does it mean the prebuilt is also working on Electron? If yes, I guess we need a config file  to tell the prebuilt path. 
I found https://github.com/serialport/bindings-cpp/commit/16f966233930bc7c7302d2b7a53d70282b42e165 and can't see the same thing after binding was moved to bindings-cpp.

andrewrt, 2022-09-14:
> @mimamuh you saved me mate! Following the steps in the [docs you linked](https://electron-react-boilerplate.js.org/docs/native-modules/) regarding how native modules are handled in [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate) set me on the right path.
> 
> I'll outline the steps I took to solve my `No native build was found for platform` error. I've modeled my project's webpack configs on the ones in [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate), so these steps may be helpful for those using [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate) or those with a similar webpack setup:
> 
>     1. `externals: ['serialport']` in webpack config -> in my case I just added it right to `webpack.config.main.prod` since I'm not merging configs like in [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate)
> 
>     2. `library: { type: 'commonjs2' }` in webpack config's `output`
> 
>     3. (if you've already installed `serialport`, `npm uninstall serialport` in your root `package.json`)
> 
>     4. `cd ./release/app`  then `npm install serialport`
> 
> 
>     * (https://electron-react-boilerplate.js.org/docs/native-modules/#native-modules-in-electron-react-boilerplate)
> 
>     * (https://github.com/electron-react-boilerplate/electron-react-boilerplate/blob/main/.erb/scripts/check-native-dep.js)
> 
> 
> If you're using [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate) you should be all set, they handle module linking in the `./release/app/package.json`: https://github.com/electron-react-boilerplate/electron-react-boilerplate/blob/main/release/app/package.json
> 
> In my case I set `serialport: =10.4.0` in both the `./release/app/package.json` and `devDependencies` in my root `package.json` since I'm using an npm workspace monorepo and not the exact folder structure of [electron-react-boilerplate](https://github.com/electron-react-boilerplate/electron-react-boilerplate). If anyone has a better solution to this I would welcome it but won't pursue it further here since it is outside the scope of `serialport`.
> 
> Hope this helps, thanks everyone!

This was a good guide - it seems like the latest ERB has the following line in the `webpack.config.base.ts`:
```js

import { dependencies as externals } from '../../release/app/package.json';

const configuration: webpack.Configuration = {
  externals: [...Object.keys(externals || {})],
```
I stumbled across this thread because I'm still seeing the serialport binding issues despite the externals being already included. Doesn't change if i comment out the above and explicitly set serialport as the only external. 

Output: 
```sh

An unhandled error occurred inside electron-rebuild
node-gyp failed to rebuild '/Users/andrew/code/js/electron/my-cool-app/release/app/node_modules/@serialport/bindings-cpp'.
For more information, rerun with the DEBUG environment variable set to "electron-rebuild".

Error: `make` failed with exit code: 2

Error: node-gyp failed to rebuild '/Users/andrew/code/js/electron/my-cool-app/release/app/node_modules/@serialport/bindings-cpp'.
For more information, rerun with the DEBUG environment variable set to "electron-rebuild".

Error: `make` failed with exit code: 2

    at NodeGyp.rebuildModule (/Users/andrew/code/js/electron/my-cool-app/node_modules/electron-rebuild/lib/src/module-type/node-gyp.js:120:19)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async ModuleRebuilder.rebuildNodeGypModule (/Users/andrew/code/js/electron/my-cool-app/node_modules/electron-rebuild/lib/src/module-rebuilder.js:98:9)
    at async ModuleRebuilder.rebuild (/Users/andrew/code/js/electron/my-cool-app/node_modules/electron-rebuild/lib/src/module-rebuilder.js:128:14)
    at async Rebuilder.rebuildModuleAt (/Users/andrew/code/js/electron/my-cool-app/node_modules/electron-rebuild/lib/src/rebuild.js:149:13)
    at async Rebuilder.rebuild (/Users/andrew/code/js/electron/my-cool-app/node_modules/electron-rebuild/lib/src/rebuild.js:112:17)
    at async /Users/andrew/code/js/electron/my-cool-app/node_modules/electron-rebuild/lib/src/cli.js:158:9
Error: Command failed: ../../node_modules/.bin/electron-rebuild --force --types prod,dev,optional --module-dir .
    at checkExecSyncError (node:child_process:841:11)
    at execSync (node:child_process:912:15)
    at Object.<anonymous> (/Users/andrew/code/js/electron/my-cool-app/.erb/scripts/electron-rebuild.js:16:11)
    at Module._compile (node:internal/modules/cjs/loader:1126:14)
    at Module.m._compile (/Users/andrew/code/js/electron/my-cool-app/node_modules/ts-node/src/index.ts:1618:23)
    at Module._extensions..js (node:internal/modules/cjs/loader:1180:10)
    at Object.require.extensions.<computed> [as .js] (/Users/andrew/code/js/electron/my-cool-app/node_modules/ts-node/src/index.ts:1621:12)
    at Module.load (node:internal/modules/cjs/loader:1004:32)
    at Function.Module._load (node:internal/modules/cjs/loader:839:12) {
  status: 255,
```
...

petertorelli, 2022-10-08:
Same issue here, but I'm using the electron/electron-forge boilerplate, not the react one. This means there is only the `.webpack/{main|renderer}` folders after `yarn start`.

@andrewrt @mimamuh Is there a hack to make this work for non react boilerplates?

NoahAndrews, 2022-10-21:
@petertorelli While it's hardly elegant, this note from @chetbox works for me (with ` externals: { serialport: 'serialport' }` in `webpack.main.config.js`):

> With Electron Forge I can get a packaged (non-ASAR) build working by copying `serialport` and its dependencies into the packaged `electron-forge` folder.
>
> ```
> yarn package
> cp -r [...]

jvolker, 2022-12-01:
@chetbox @NoahAndrews How would this work with Electron Forge during development when there is no `out` directory created yet?

I added this to `webpack.renderer.config.js`
```
externals: {
   serialport: "serialport",
},
```

And I'm still stuck with: `Uncaught ReferenceError: serialport is not defined`

Thanks.

NoahAndrews, 2022-12-01:
@jvolker I actually meant to come clarify what we're _actually_ doing, which is only based on @chetbox's solution. Thanks for the reminder. @petertorelli, maybe this is useful to you too.

We have a `copy-to-package-directory.js` file:
```js
const fs = require('fs');
const path = require('path');

// https://stackoverflow.com/a/22185855/4651874
function copyRecursiveSync(src, dest) {
    let exists = fs.existsSync(src);
    let stats = exists && fs.statSync(src);
    let isDirectory = exists && stats.isDirectory();
    if (isDirectory) {
        console.log(`Making directory ${dest}`);
        fs.mkdirSync(dest, { recursive: true });
        fs.readdirSync(src).forEach(function(childItemName) {
            console.log(`Copying ${childItemName}`);
            copyRecursiveSync(path.join(src, childItemName),
                path.join(dest, childItemName));
        });
    } else {
        console.log(`Copying ${src} to ${dest}`);
        fs.copyFileSync(src, dest);
    }
}

// Squirrel won't package up unexpected files unless they are in the "resources" folder
// See https://github.com/electron-userland/electron-forge/issues/135#issuecomment-991376807
module.exports = function(extractPath, electronVersion, platform, arch, done) {
    // https://github.com/serialport/node-serialport/issues/2464#issuecomment-1122454950
    copyRecursiveSync(path.join('node_modules', 'serialport'), path.join(extractPath, 'resources', 'app', 'node_modules', 'serialport'));
    copyRecursiveSync(path.join('node_modules', '@serialport'), path.join(extractPath, 'resources', 'app', 'node_modules', '@serialport'));
    copyRecursiveSync(path.join('node_modules', 'debug'), path.join(extractPath, 'resources', 'app', 'node_modules', 'debug'));
    copyRecursiveSync(path.join('node_modules', 'ms'), path.join(extractPath, 'resources', 'app', 'node_modules', 'ms'));
    copyRecursiveSync(path.join('node_modules', 'node-gyp-build'), path.join(extractPath, 'resources', 'app', 'node_modules', 'node-gyp-build'));

    console.log("Done copying files");
    done();
}
```

In `package.json`, we configure Forge to run that script:
```json
"forge": {
  "packagerConfig": {
    "afterExtract": [
      "copy-to-package-directory.js"
    ]
  }
}
```

jvolker, 2022-12-02:
Thanks, @NoahAndrews. 

Unfortunately, it doesn't solve the issue during development though, it seems. The "afterExtract" hook is not being triggered then.

jvolker, 2022-12-07:
Thanks. Yes, I can confirm this works using serialport in the main process. Is there any chance to get this working in renderer as well?

petertorelli, 2022-12-24:
@jvolker & @NoahAndrews ... the simple fix of `exclude`-ing it solves my problem, thank you! however I currently cannot test in a packaged prod environment because I can't seem to get electron-forge to package properly: my other [issue](https://github.com/electron/forge/issues/3120). Ty!

petertorelli, 2022-12-26:
@jvolker - While it might work in this instance, that's waaaay too gnarly a hack to carry forward in a shipping project. I'm trying to migrate an Electron app we've been shipping since 2018 to Mac m1. Our "ancient" build environment is still stable so we continue to use it (electron 10 and builder 23) and just punt on m1 support for now (and we can't rely on rosetta due to libusb bandwidth issues). Ideally we're forcing electron-forge to build non-web apps for hardware front-ends (USB, VISA, serial), which is probably a bad idea to force a square peg into a round hole. What we need is a boilerplate packager that is designed for non-web hardware interfacing apps that use electron, and jettison the web stuff! :)

glenn-kroeze, 2023-01-06:
I was struggling with a `No native build was found for platform` error message when importing `serialport` in Electron's main process. I was using `electron-forge` to build my distributable. I also tried using `electron-rebuild` manually, to no avail.

I spent a couple of days digging through Stack Overflow posts and GitHub issues and did some experiments. In the end it looks like the issue is caused by `electron-rebuild` not deeming `serialport`'s native node modules necessary to be rebuilt on certain platforms. In my case, `electron-rebuild` was rebuilding the `serialport` native node modules on my M1 MacBook Pro (arm64), but wasn't rebuilding these modules on a Raspberry Pi 4 Model B (armv7l).

What fixed it for me in the end (credit to the [this][1] GitHub comment) is running the following commands before Electron's packaging step:
```
rm --recursive node_modules/@serialport
npm install --no-save --build-from-source serialport@10.4.0
```

This will first delete all the contents of your `node_modules/@serialport` folder, which contains prebuilt native node modules for several platforms in the `bindings-cpp` package. Next, we build `serialport`'s native node modules from scratch, so that we (hopefully) end up with a native node module that matches our current platform. 

Make sure the version of `serialport` used in the above command matches the version you depend on!

  [1]: https://github.com/serialport/node-serialport/issues/2463#issuecomment-1227334940

petertorelli, 2023-01-06:
@glenn-kroeze I'm using serialport from within a fork started by the renderer process (my app can run with or without a renderer, so the same fork can be called from within either, depending on whether a window is created. I support GUI and CLI modes for CI/CD.). This creates all kinds of problems in forge/webpack. I don't think this is a serialport issue at all, but a packager issue.

rtoscani, 2023-02-24:
After struggling a lot with this issue, i was able to resolve using another approach. In my case we need to use `usb` and `serialport` dependencies.

First add `externals: ['usb', 'serialport'],` to `webpack.main.config.ts`

Then inside `forge.config.ts` we added a hook to include these modules on `node_modules` internal dependencies

```javascript
hooks: {
    packageAfterPrune: async (forgeConfig, buildPath) => {
        console.log(buildPath);

        const packageJson = JSON.parse(fs.readFileSync(path.resolve(buildPath, 'package.json')).toString());

        packageJson.dependencies = {
            serialport: '^10.5.0',
            usb: '^2.8.0',
        };

        fs.writeFileSync(path.resolve(buildPath, 'package.json'), JSON.stringify(packageJson));

        return new Promise((resolve, reject) => {
            const npmInstall = spawn('yarn', ['install', '--production=true'], {
                cwd: buildPath,
                stdio: 'inherit',
                shell: true,
            });

            npmInstall.on('close', code => {
                if (code === 0) {
                    resolve();
                } else {
                    reject(new Error('process finished with error code ' + code));
                }
            });

            npmInstall.on('error', error => {
                reject(error);
            });
        });
    },
},
```

Not an elegant solution but works. We cannot use references from original `package.json` because these libraries are used from another internal library.

thehans, 2023-04-20:
Here is the hook snippet I was pointed to on discord, which worked for me

```js
// forge.config.js

...

  hooks: {
    packageAfterPrune: async (_, buildPath, __, platform) => {
      const commands = [
        "install",
        "--no-package-lock",
        "--no-save",
        "serialport",
      ];

      return new Promise((resolve, reject) => {
        const oldPckgJson = path.join(buildPath, "package.json");
        const newPckgJson = path.join(buildPath, "_package.json");

        fs.renameSync(oldPckgJson, newPckgJson);

        const npmInstall = spawn("npm", commands, {
          cwd: buildPath,
          stdio: "inherit",
          shell: true,
        });

        npmInstall.on("close", (code) => {
          if (code === 0) {
            fs.renameSync(newPckgJson, oldPckgJson);

            /**
             * On windows code signing fails for ARM binaries etc.,
             * we remove them here
             */
            if (platform === "win32") {
              const problematicPaths = [
                "android-arm",
                "android-arm64",
                "darwin-x64+arm64",
                "linux-arm",
                "linux-arm64",
                "linux-x64",
              ];

              problematicPaths.forEach((binaryFolder) => {
                fs.rmSync(
                  path.join(
                    buildPath,
                    "node_modules",
                    "@serialport",
                    "bindings-cpp",
                    "prebuilds",
                    binaryFolder
                  ),
                  { recursive: true, force: true }
                );
              });
            }

            resolve();
          } else {
            reject(new Error("process finished with error code " + code));
          }
        });

        npmInstall.on("error", (error) => {
          reject(error);
        });
      });
    },
  },
```

And then in `webpack.main.config.js`, just:
```js
externals: { serialport: "serialport" },
```

deyemiobaa, 2023-04-24:
@mimamuh I'm using the electron-react-boilerplate and I intend to use the data in one of my components. Could you please help me confirm if I'm using it the right way?

In `main.ts`
```ts
...
  mainWindow = new BrowserWindow({
    show: false,
    width: 800,
    height: 480,
    icon: getAssetPath('icon.png'),
    webPreferences: {
      preload: app.isPackaged
        ? path.join(__dirname, 'preload.js')
        : path.join(__dirname, '../../.erb/dll/preload.js'),
      nodeIntegration: true,
    },
    autoHideMenuBar: true,
    titleBarStyle: 'hidden',
    frame: false,
  });

  const port = new SerialPort({ path: '/dev/ttyAMA0', baudRate: 115200 });
  const parser = port.pipe(new ReadlineParser({ delimiter: '\r\n' }));
  if (port.isOpen) {
    parser.on('data', (data: string) => {
      if (data.startsWith('$')) {
        mainWindow?.webContents.send('serial-data', data);
      }
    });
  }

...

```

In `preload.ts`

```ts
...
const electronHandler = {
  ipcRenderer: {
    sendMessage(channel: Channels, args: unknown[]) {
      ipcRenderer.send(channel, args);
    },
    on(channel: Channels, func: (...args: unknown[]) => void) {
      const subscription = (_event: IpcRendererEvent, ...args: unknown[]) =>
        func(...args);
      ipcRenderer.on(channel, subscription);

      return () => {
        ipcRenderer.removeListener(channel, subscription);
      };
    },
    once(channel: Channels, func: (...args: unknown[]) => void) {
      ipcRenderer.once(channel, (_event, ...args) => func(...args));
    },
  },
  serialData: (callback: any) => ipcRenderer.on('serial-data', callback),
...
```

In the component
```jsx
  useEffect(() => {
    window.electron.serialData((_event: any, data: string) => {
      setText(data);
    });
  }, []);
```

tkw722, 2023-07-24:
So, upon further investigation, I've discovered why node-gyp-build fails to find `bindings.node`. It boils down to this line in **bindings-cpp/dist/load-bindings.js**:

`const binding = (0, node_gyp_build_1.default)((0, path_1.join)(__dirname, '../'));`

In this case, **__dirname**, is set to the root of the .webpack folder rather than the root of **node_modules/serialport/node_modules/@serialport/bindings-cpp/**. This leads to the checks for `bindings.node` failing leading to the error I mentioned previously. I still need to track down where/when **__dirname** is set, but that's what appears to be going awry, at least for me.

tkw722, 2023-07-24:
I'm making progress, I believe.  I figure I'll keep this updated as I do in hopes it will help someone else out facing this same problem. Turns out webpack has a **__dirname** polyfill. Without that turned on, __dirname will not behave predictably it seems in the current webpack. I've added:

`node: {
    __dirname: true
  },`

to **webpack.main.config.ts** which seems to be moving things forward. I can now verify that __dirname is behaving as would be expected. Now I'm facing a new error:

`TypeError: Cannot read properties of undefined (reading 'SerialPort')`

So it seems like it was finally able to find the native module correctly. Progress is a new error! ;-)

tkw722, 2023-07-24:
Wow. I think I might cry. I have it working.

Somewhere in the misery of working on this, SerialPort was being included incorrectly. I had:

`import SerialPort from 'serialport';`

Which I changed back to:

`import { SerialPort } from 'serialport';`

Which matches the SerialPort documentation. I think the prior import was a vestige of hacking around with my initial attempt at loading SerialPort in the renderer process and experimenting there.

So the punchline to all of my debugging and the solution to at least my situation was:

**ADD**

`node: { __dirname: true },`

**TO**

`webpack.main.config.ts`

GazHank, 2023-07-24:
Hi @tkw722 I've not used electron forge much myself, but I believe that the [bazecor](https://github.com/Dygmalab/Bazecor) project recently transitions to electron forge from electron builder.  I wonder if any of those changes would be worth reviewing to see if it gives any pointers for problems you are seeing with the prod build

javierguzman, 2023-07-25:
Thanks for sharing your findings @tkw722!  I do not recall very well (short memory guy here) but for development I just had to do eval(require(blabla)) and that is...Then for production/bundling I had to do the one of the several after prune hook solution if I am not mistaken

Burzo, 2023-09-05:
> Here is the hook snippet I was pointed to on discord, which worked for me
>
> ```js
> // forge.config.js
>
> ...
>
>   hooks: {
>     packageAfterPrune: async (_, buildPath, __, platform) => {
>       const [...]

Oh hey, my Discord answer made it to GitHub!

I managed to get an Electron Forge + Serialport app built and signed for MacOS (arm and intel), Windows, and Linux so if anyone is stuck or needs help hit me up on discord - `.burzo`. Hopefully I'll be able to help.

andrey-pr, 2024-01-01:
> For react-electron-boilerplate:
> 
> 1. Add `externals: ['serialport']` in `webpack.config.main.prod`
> 
> 2. `library: { type: 'commonjs2' }` in webpack config's `output` in `webpack.config.main.prod`
> 
> 3. `cd ./release/app`  then `npm install serialport`

Thank you @sh0shinsha, it helped

EDais, 2024-02-27:
In case it helps anyone else out who also gets directed to this thread for help with compiling a Node application using serialport using pkg but without React, Electron or Webpack, this is what finally worked for my setup.

Add this to `package.json`:

```json
"pkg": {
    "assets": [ "node_modules/@serialport/bindings-cpp/prebuilds/**/*" ]
}
```

Adjust the path based on the relative location of your `prebuilds` directory, and optionally include only the platform/architecture(s) you're targeting, but this should package the native binaries in the virtual file system in a way that `node-gyp` can load them.

AviadCohen24, 2024-03-04:
> > For react-electron-boilerplate:
> > 
> > 1. Add `externals: ['serialport']` in `webpack.config.main.prod`
> > 2. `library: { type: 'commonjs2' }` in webpack config's `output` in `webpack.config.main.prod`
> > 3. `cd ./release/app`  then `npm install serialport`

Thank you @sh0shinsha, it helped

Astrojoinl, 2024-06-28:
the problem is to make it work in the rederer process, where webpack doesn't seem to help to have this functionality that for example can be seen here [https://github.com/serialport/electron-serialport/tree/master](url), i'm using `electron-forge` and the webpack renderer system seems to have no way to function properly:

>  ERROR in ./node_modules/@serialport/bindings-cpp/dist/linux-list.js 7:22-46
  Module not found: Error: Can't resolve 'child_process' in '(...)\node_modules\@serialport\bindings-cpp\dist'

shadesoma, 2024-09-20:
Do not import the serialport in the renderer file, call it in index.ts after creating the window.

rubenswebdev, 2024-10-18:
> In case it helps anyone else out who also gets directed to this thread for help with compiling a Node application using serialport using pkg but without React, Electron or Webpack, this is what [...]

Thanks!!! this help me a lot!

Haavard15, 2025-04-14:
Hi everyone,

I encountered significant issues getting native Node.js modules (like @serialport/bindings-cpp, node-hid) correctly packaged and signed on macOS using Electron Forge, leading to errors like:

dlopen errors mentioning invalid code signatures or missing Team ID.
The final .app being reported as "damaged and can't be opened".
codesign --verify failing with "a sealed resource is missing or invalid" pointing to the .node files.
The core problem seems to be the timing of code signing. Electron Forge signs the application bundle as one of its final steps. If native modules are modified (e.g., by manually signing them) after the main application bundle has been signed, it breaks the seal and invalidates the overall signature. Standard hooks like postPackage run too late, and other hooks like packageAfterCopy might run too early (before the modules are installed/rebuilt in the temporary directory).

Solution:

The solution involves manually signing the required .node files within the packageAfterPrune hook, immediately after they have been installed and rebuilt for Electron, but before the hook finishes (resolves its promise) and Electron Forge proceeds with its own packaging and signing steps.

Here's the relevant snippet from my working forge.config.ts:

```
import path from 'path';
import * as fs from 'fs-extra';
import { spawn } from 'child_process';
import type { ForgeConfig } from '@electron-forge/shared-types';

// Helper function to run codesign
const runCodeSign = (filePath: string): Promise<void> => {
  return new Promise((resolve, reject) => {
    console.log(`Manually signing: ${filePath}`);
    // --- Replace with your actual Developer ID Application identity ---
    const identity = 'Developer ID Application: Your Name (TEAMID)';
    // Add --options=runtime if using hardened runtime
    const args = ['--sign', identity, '--force', '--deep', '--options=runtime', filePath];
    const cs = spawn('codesign', args, { stdio: 'inherit' });

    cs.on('close', (code) => {
      if (code === 0) {
        console.log(`Successfully signed: ${filePath}`);
        resolve();
      } else {
        console.error(`codesign failed for ${filePath} with code ${code}`);
        reject(new Error(`codesign failed for ${filePath} with code ${code}`));
      }
    });
    cs.on('error', (err) => {
      console.error(`Failed to start codesign for ${filePath}:`, err);
      reject(err);
    });
  });
};

const config: ForgeConfig = {
  // ... other packagerConfig options (asar, icon, etc.)
  packagerConfig: {
    asar: true,
    // ...
    osxSign: {
      // --- Ensure your identity is set here too ---
      identity: 'Developer ID Application: Your Name (TEAMID)',
      // entitlements: 'path/to/entitlements.plist', // Add if needed
      // entitlementsInherit: 'path/to/entitlements.inherit.plist', // Add if needed
    },
    // ... osxNotarize config
  },
  hooks: {
    // ... other hooks like readPackageJson might still be needed ...

    packageAfterPrune: async (forgeConfig, buildPath) => {
      console.log('Starting npm install & rebuild for native modules after prune...');
      // --- List ALL native modules that need rebuilding/signing ---
      const nativeModules = ['serialport', 'node-hid' /*, ... other modules */];

      // --- Choose ONE strategy for installing native modules ---

[... 61 lines cut ...]
              console.log('Not macOS, skipping native module signing.');
              resolve(); // Resolve if not macOS
            }
            // ---- END SIGNING LOGIC ----
          });

          rebuild.on('error', (rebuildError) => {
            console.error('electron-rebuild spawn error:', rebuildError);
            reject(rebuildError);
          });

        } catch (error) {
          console.error('Error during packageAfterPrune hook:', error);
          reject(error);
        }
      });
    },

    // ... postMake hook etc. ...
  },
  // ... makers, plugins ...
};

export default config;
```