serialport/node-serialport #2493: parser.on("data")  get Nothing

zhangbowei, 2022-05-06:
### SerialPort Version

new

### Node Version

10.24.1

### What steps will reproduce the bug?

const { SerialPort } = require('serialport')
const port = new SerialPort({ path: '/dev/ttySAC1', baudRate: 115200 })
const { DelimiterParser } = require('@serialport/parser-delimiter')
const parser = port.pipe(new DelimiterParser({ delimiter: '$$$' }))

process.on('message', (message) => {
console.log('test")
})

parser.on('data', function (res) {
       console.log("666")
})

### What happens?

when port get data, there is nothing.
But when process.on get data, eg. "test",
I will get all all "666", It depends on how long I wait for a "test"

### What should have happened?

"666" should not wait for "test"
their are no relative.

reconbot, 2022-05-06:
How does process get a message event?

zhangbowei, 2022-05-07:
const mqtt = require('mqtt')
const ID = "295003";
const host = ''
const mPort = '1883'
const clientId = `mqtt_${Math.random().toString(16).slice(3)}`
const OPTIONS = {
    clientId,
    clean: true,
    connectTimeout: 4000,
    username: 'test',
    password: 'test',
    reconnectPeriod: 1000,
}
client.on('message', (topic, message) => {
    process.send("mqtt$$$")
})

process.on('message', (message) => {
    console.log("233")
})

zhangbowei, 2022-05-07:
233 doesn't  impact mqtt$$$
but  test block 666.
If I remove processs in /6qUart.js，
"666" will come right.
eg.
const { SerialPort } = require('serialport')
const port = new SerialPort({ path: '/dev/ttySAC1', baudRate: 115200 })
const { DelimiterParser } = require('@serialport/parser-delimiter')
const parser = port.pipe(new DelimiterParser({ delimiter: '$$$' }))

/*process.on('message', (message) => {
console.log('test")
})*/

parser.on('data', function (res) {
console.log("666")
})

zhangbowei, 2022-05-07:
I guess because of process.nextTick and setTimeout.
What should I do？

zhangbowei, 2022-05-10:
beacuse of node version
armv7l node v10.24.1 => armv7l node v12.22.12  
solve it.

closed 2022-05-10 by zhangbowei