serialport/node-serialport #2508: Unable to set LowLatency while this is documented.

BartHertog, 2022-06-03:
In the documentation about setting the [options](https://serialport.io/docs/api-stream#set) for a port the LowLatency option is mentioned:

` * {Boolean} [setOptions.lowLatency=true] enables low latency mode (linux specific - admin rights may be required)`

Unfortunately when looking in the typescript code (V10.4.0) there is no mention of the latency option:
```
declare namespace SerialPort {
    interface SetOptions {
        brk?: boolean;
        cts?: boolean;
        dsr?: boolean;
        dtr?: boolean;
        rts?: boolean;
    }
}
```
We are working on a Linux platform and would like to enable LowLatency from typescript. Looking at the bindings-cpp code this should be possible. So for our case updating the typescript code would be best but up until that point maybe update the docs?

Best regards,

Bart

richdunajewski, 2022-06-04:
@BartHertog I just ran into the same issue. It looks like there is a bindings-specific interface called `LinuxSetOptions` which extends `SetOptions`, but the issue is `SerialPort.set()` is expecting an argument of type `SetOptions`.

However, if you inspect the SerialPort object after opening the port, you'll see there is a `port` property which is of type `LinuxPort` and that has a `set()` method that accepts `LinuxSetOptions`.

So for example, change:

```javascript
const port = new SerialPort(...);
port.set({lowLatency: true});
```

to

```javascript
const port = new SerialPort(...);
port.port.set({lowLatency: true});
```

BartHertog, 2022-06-04:
Hello @richdunajewski,

Thank you for this tip! I will look into this next week.

Best regards,

Bart

BartHertog, 2022-06-10:
Hello @richdunajewski,

Currently we are using version 9.4.0. In this version it is not possible to do this. Are you using a newer version?

We have looked in to using `SerialPort.Binding.set()` however the `SetOptions` interface just does not define the lowlatency option.