serialport/node-serialport #2527: .close NOT WORKING !!! Max ram memory usage

Fadeusz, 2022-07-31:
### SerialPort Version

10.4.0

### Node Version

v16.16.0

### Platform

Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

### Architecture

ARM

### What steps will reproduce the bug?

https://pastebin.com/ZwBSUScU - SerialPort.ts

-------

import { SerialPort } from './App/SerialPort';

SerialPort.list().then (
  ports => ports.forEach(port =>SerialPort.init(port)),
  err => console.log(err)
)

#^ run.ts

### What happens?

message written
message written
<Buffer 41 54 0d 0a 4f 4b 0d 0a>
connection manualy closed
connection manualy closed

<--- Last few GCs --->

[29870:0x40ee428]   199884 ms: Mark-sweep 225.8 (236.8) -> 223.9 (237.0) MB, 3217.9 / 0.0 ms  (average mu = 0.177, current mu = 0.032) task scavenge might not succeed
[29870:0x40ee428]   204204 ms: Mark-sweep 225.9 (237.0) -> 224.0 (236.8) MB, 4140.8 / 0.0 ms  (average mu = 0.107, current mu = 0.041) task scavenge might not succeed
[29870:0x40ee428]   208464 ms: Mark-sweep 226.4 (236.8) -> 224.1 (236.8) MB, 4152.3 / 0.0 ms  (average mu = 0.066, current mu = 0.025) task scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory

### What should have happened?

Connection to the 3 serial ports has been established. Only one of them should work and by my assumption 2 of them should close after 10 seconds. The .close function is called but nothing happens. The program starts using more and more ram until it uses the quota.

### Additional information

If I connect to only one correct port using the framework, no problem, but the program is supposed to connect to everyone on the list and check them. Unfortunately, the close function doesn't work if I want to disconnect from them

GazHank, 2023-06-27:
Hi @karlos1998 sorry I havent had chance to take a look at this before, are you still facing issues?

MBerka, 2023-09-11:
Failure to close should not automatically mean running out of memory. The code does not show which ports were opened; if any was `/dev/ttyAMA0` or another port that Raspberry Pi sometimes uses for Bluetooth, then the crashes could be due to the interference between the program and the Bluetooth service. I have seen the same error messages on a Pi 3B+ resolved by `systemctl disable bluetooth` followed by a reboot.