serialport/node-serialport #2531: Electron builder Windows app built in Docker or Mac Fails

jpakulski, 2022-08-18:
### SerialPort Version

10.4.0

### Node Version

16.14.1

### Electron Version

20.0.2

### Platform

Mac / Docker image based on electronuserland/builder:wine

### Architecture

x64

### Hardware or chipset of serialport

N/A

### What steps will reproduce the bug?

Clone the serial port demo project:
git clone https://github.com/serialport/electron-serialport.git

Add electron-builder config to package.json (test config below):

```json
"build": {
    "appId": "org.xxx.serialtest",
    "asar": true,
    "productName": "SerialTest",
    "buildDependenciesFromSource": false,
    "directories": {
      "output": "builder-output"
    },
    "win": {
      "target": "nsis"
    },
    "nsis": {
      "oneClick": true,
      "runAfterFinish": true,
      "allowToChangeInstallationDirectory": false
    }
  }
```

Add the following scripts to package,json:
```json
"scripts": {
    "clean": "rm -rf builder-output",
    "dist:win": "electron-builder --win -p never"
  },
```

Execute the following (on Mac or Docker (electronuserland/builder:wine) container):
npm i
npm run clean
npm run dist:win

### What happens?

The build seems to work:

```
  • electron-builder  version=23.3.3 os=21.5.0
  • loaded configuration  file=package.json ("build" field)
  • writing effective config  file=builder-output/builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=@serialport/bindings-cpp@10.6.3 platform=win32 arch=x64
  • packaging       platform=win32 arch=x64 electron=17.4.11 appOutDir=builder-output/win-unpacked
  • default Electron icon is used  reason=application icon is not set
  • building        target=nsis file=builder-output/SerialTest Setup 1.0.2.exe archs=x64 oneClick=true perMachine=false
  • building block map  blockMapFile=builder-output/SerialTest Setup 1.0.2.exe.blockmap
```

But when the app is run on Windows, I get the following error in the dev tools:

```
Uncaught Error: \\?\C:\Users\j\AppData\Local\Temp\643069be-b2d9-4e55-af7e-95fffc6f6399.tmp.node is not a valid Win32 application.
\\?\C:\Users\j\AppData\Local\Temp\643069be-b2d9-4e55-af7e-95fffc6f6399.tmp.node
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:5:1812)
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1199:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:5:2039)
    at Module.load (node:internal/modules/cjs/loader:988:32)
    at Module._load (node:internal/modules/cjs/loader:829:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13343)
    at Function.o._load (node:electron/js2c/renderer_init:33:356)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at load (C:\Users\j\AppData\Local\Programs\electron-serialport\resources\app.asar\node_modules\node-gyp-build\index.js:22:10)
```

I'm sure I'm missing a step or just don't understand how to use/configure the prebuilt versions or how the Node-API versions are supposed to work.

Everything was working in our CI env. (Docker build) before updating to Serialport >= 10.

### What should have happened?

We should get a working application as before updating Serialport to version 10+.

### Additional information

I tried this with a rainbow of node versions / electron versions / serial port versions all resulting in the same error.

Back in the pre Node-API days this line:
```
rebuilding native dependencies  dependencies=@serialport/bindings-cpp@10.6.3 platform=win32 arch=x64
```

was always followed by:
```
install prebuilt binary  name=@serialport/bindings version=9.0.3 platform=win32 arch=x64
```

I guess native dependancies are being built for the wrong platform rather than the prebuilt ones being used. There is an autoDetect function which is supposed to supply the correct bindings but somehow it does not work?

Any help is super appreciated.

Cheers.

jpakulski, 2022-08-18:
Probably related to: https://github.com/serialport/node-serialport/issues/2463

GazHank, 2023-06-27:
per https://github.com/serialport/node-serialport/issues/2619 cross building using electron-builder should now work correctly in electron-builder 24.5.1