serialport/node-serialport #2534: Can't install on Raspberry Pi 4

fingerskier, 2022-08-26:
### SerialPort Version

10.3

### Node Version

16.13

### Electron Version

17.0

### Platform

Linux raspberrypi 5.15.56-v7l+ #1575 SMP Fri Jul 22 20:29:46 BST 2022 armv7l GNU/Linux

### Architecture

ARM

### What steps will reproduce the bug?

Clone the `electron-serialport` repo, then `npm i`, then open developer tools to see the following output.

### What happens?

Uncaught Error: No native build was found for platform=linux arch=arm runtime=electron abi=101 uv=1 armv=default libc=glibc node=16.13.0 electron=17.0.0
    loaded from: /home/spectrum/dev/electron-serialport/node_modules/@serialport/bindings-cpp

    at Function.load.path (/home/spectrum/dev/electron-serialport/node_modules/node-gyp-build/index.js:59:9)
    at load (/home/spectrum/dev/electron-serialport/node_modules/node-gyp-build/index.js:21:30)
    at Object.<anonymous> (/home/spectrum/dev/electron-serialport/node_modules/@serialport/bindings-cpp/dist/load-bindings.js:10:46)
    at Object.<anonymous> (/home/spectrum/dev/electron-serialport/node_modules/@serialport/bindings-cpp/dist/load-bindings.js:23:3)
    at Module._compile (node:internal/modules/cjs/loader:1116:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1169:10)
    at Module.load (node:internal/modules/cjs/loader:988:32)
    at Module._load (node:internal/modules/cjs/loader:829:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Function.o._load (node:electron/js2c/renderer_init:33:356)

### What should have happened?

Expect it to load without error.

v0na, 2022-09-10:
I found a way to solve the issue by spending 2 days.

Install node gyp packages globally;
`sudo npm install -g node-gyp node-pre-gyp`

Enter the serial port bindings cpp dir.
`cd node_modules/@serialport/bindings-cpp`

`run the command to compile module`
sudo node-gyp configure build

I hope it helps.

fingerskier, 2022-09-29:
Thank you so much!  That did the trick...

closed 2022-09-29 by fingerskier