serialport/node-serialport #2540: Unhandled C++ Exception / 0x7ff8f8034fd9

hrueger, 2022-09-24:
### SerialPort Version

10.4.0

### Node Version

16.13.2

### Electron Version

20.1.4

### Platform

Microsoft Windows NT 10.0.19044.0 x64

### Architecture

x64

### What steps will reproduce the bug?

I unfortunately don't know. This happens occasionally in a complex application and I don't have found any way to trigger the bug. It just happens sometimes.

### What happens?

The native code crashes. See here for the sentry stack trace: https://sentry.internal.makepro-x.com/share/issue/00c398c251ff426db2063a206d9117a1/

I know that this error comes from `node-serialport`, because if I open [37a99351-6a52-4763-a2d4-6ab1040c4070.tmp.node](https://github.com/serialport/node-serialport/files/9638710/37a99351-6a52-4763-a2d4-6ab1040c4070.tmp.node.zip) in a text editor, I find ascii text like `node-serialport:DrainBaton` or `D:\a\bindings-cpp\bindings-cpp\src\serialport`.

Unfortunately, I don't have more information available. Is there a place where I could get the .pdb files in order to upload them to sentry?

### What should have happened?

It should not crash :-)

hrueger, 2022-09-24:
Could this be the problem?
[Electron and the V8 Memory Cage](https://www.electronjs.org/blog/v8-memory-cage)
see also https://github.com/justinlatimer/node-midi/pull/228#issuecomment-1256993570

Julusian, 2022-09-24:
Sharing info over here too, the memory cage is enabled in electron 21 not 20, so wont be at fault here.
I did look over the native code, and didnt see the broken pattern, so I think this module should be ok.

Additionally, when hitting the memory cage issue it is very clear why it crashed:
```
[824319:0924/182105.427206:ERROR:node_bindings.cc(146)] Fatal error in V8: v8_ArrayBuffer_NewBackingStore When the V8 Sandbox is enabled, ArrayBuffer backing stores must be allocated inside the sandbox address space. Please use an appropriate ArrayBuffer::Allocator to allocate these buffers.
```