serialport/node-serialport #2542: Serialport package issue after electron build with electron-builder

megha24051994, 2022-09-29:
### SerialPort Version

10.0.0

### Node Version

14.19.0

### Electron Version

19.0.0

### Platform

x64

### What steps will reproduce the bug?

Using serialport in an electron-ionic project in the renderer side. Serialport package is installed and loaded into the project as follows in the serialport service constructor:

`this.serialport = window.require('serialport');`

The project works and serialport list function works when the electron window is opened using the command `npx cap open @capacitor-community/electron`. 

There are 2 issues afterwhich:
1. The port creation doesn't work and but doesn't throw any error to analyse the issue. Also tried adding it with a catch statement.
`this.port= new this.serialport .SerialPort(serialDevice.name, { baudRate: 9600, autoOpen: false });`
2. After building the electron project using electron-builder, the serialport module isn't detected anymore and throws the error 'Cannot find module serialport'.
[image]

### What happens?

In an electron-ionic project, the serialport package (v10.0.0) works when the electron window is opened in Dev mode using 'npx cap open @capacitor-community/electron' but it throws an error "Cannot find module 'serialport'" when the project is build using electron-builder.

### What should have happened?

Serialport package should work with the build exe as well.

Dardin-dale, 2023-02-10:
I recommend putting Serialport on the main process and not the renderer process. You can use [Electron's IPC](https://www.electronjs.org/docs/latest/tutorial/ipc) handlers to call your serialport interface from the renderer process.

surafelbm, 2025-01-27:
@megha24051994 , did you ever fix this issue please? running into the same issue

MeikMueller, 2025-01-28:
Well we never fixed it but we changed from "serialport" to "https://developer.mozilla.org/en-US/docs/Web/API/Web_Serial_API". More or less the only difference is that you have to use a Buttonpress to init it.