serialport/node-serialport #2558: Introduce MockPortBinding.writeToPort to help with testing serialport-based code

Lucas-C, 2022-11-17:
## 💥 Proposal

### What feature you'd like to see

Provide a way for `MockPortBinding` to write data to another `MockPortBinding`.

## Motivation / Pitch

Often when testing a software application based on `node-serialport`,
we have several devices (2 a least) communicating through serial ports.
It can then be useful to mock the behaviour of the program on the other end of the serial port.
This is currently difficult to do with `@serialport/binding-mock` that only provides the ability to send data to itself and/or record it.

## Pull Request

I submitted the following PR to provide this feature, including an example use case:
https://github.com/serialport/binding-mock/pull/20

reconbot, 2022-11-17:
Good ideas! I left a comment

jmbldwn, 2023-11-25:
I think what we really want is to mock the devices we connect to.  My unit tests aren't about whether I'm using SerialPort correctly as much as they are about whether my code is communicating with the external device.  To that end I need to build a test module that effectively connects to the back end of a mock serial port.  Writes to the mock port become 'data' events on my mock device.  Writes from my mock device become 'data' events on the port.

It's so close, but right now as best I can tell I can only do loop back and inspect recordings.  That doesn't let me mock the device's protocol.